MarlinFirmware/Marlin #17950: [FR] (Any possible way to attach endstop to E0 and E1 and using homing fuction on them?)

astorun, 2020-05-10:
Hi all,

I looked at the Marlin documentation and could not find anything about it.
I need to use homing functions on all extruder axis specifically on E0 and E1. I am not using it for 3d printer purposes but a special device so in my usage it needs to home to a specific position. I cannot use XYZ axis because the way of my implementation requires exclusive control on Extruder controls.

Does Marlin has any support to endstops on all E axes and ability to use homing them only on E axes?

I am not good at coding so I will really helpful if there is any forks of firmware or method to use it.

Thanks.

thisiskeithb, 2020-05-10:
[Need to ask a question? Look here first](https://github.com/MarlinFirmware/Marlin/issues/14694)

astorun, 2020-05-10:
> 
> 
> [Need to ask a question? Look here first](https://github.com/MarlinFirmware/Marlin/issues/14694)

Yep I checked them but could not find anything as I stated above there is nothing related with E Axis endstops on Marlin Documentation.

I wonder if there is anything under development or any other way todo it.

Thanks.

Bob-the-Kuhn, 2020-05-11:
As far as I know there is nothing in development.

I think everything that needs to be modified is in the **endstops.cpp** and **endstops.h** files.

The **void Endstops::init() {** and **Endstops::update()** routines will need to be modified.  The **Endstops::update()** routine is where the endstops are polled.  Use the X_min and X_max as examples to create code to poll the new endstops.

I'm not sure if **void Endstops::event_handler() {** needs to be modified.  I think that is only used when endstop interrupt mode is enabled.

You'll also need to change the following in **endstops.h** 
- line 31 `enum EndstopEnum : char {` - add the new endstops to this list
- line 44 `typedef uint16_t esbits_t` - you'll need to modify this logic so you have enough bits to keep track of the new endstops.  I suggest adding a new section that uses `typedef uint32_t esbits_t`

It's a good idea to keep customization of the configurations files to a minimum and do all the new pin definitions in the pins_YOUR_BOARD.h file.

astorun, 2020-05-12:
I ended up using X and Y axes to use the endstops. Since I am not a programmer it will probably a difficult task for me. I hope there will be an option in future to add custom amount of axes and multiple endstops for each of them just by defining them in configuration file :) Thanks though.

closed 2020-05-12 by astorun