MarlinFirmware/Marlin #17951: Extruder and Z-Axis making strange noise during operation

ComputerKing12, 2020-05-10:
**Firmware Version**
Latest bugfix-2.0 release

**Bug Description**
Whenever I go to start a print or do any motion involving the z-axis stepper motor and retracts on the extruder, I hear this odd sound that was not present before I updated the firmware.

### My Configurations

Github couldn't upload my files, so I used Google Drive instead, (link at bottom to video and zip file) 

Steps to Reproduce
1)Compile firmware
2)Flash the firmware
3)Start any print off the sd card, and listen

**Expected behavior:** [What you expect to happen]
You should hear no sound coming from your stepper motor drivers

**Actual behavior:** [What actually happens]
You hear odd sounds coming from your stepper motor drivers

#### Additional Information

t.ly/H8ei

Firmware is for Ender 3, with the BigTreeTech Mini SKR 1.3 V1.2
I also included the firmware.bin file in the Google Drive link.

swilkens, 2020-05-11:
That sounds like you are hitting the `#HYBRID_THRESHOLD` limits for those axis.

Information here:
https://marlinfw.org/docs/gcode/M913.html

Quick check is to set the limits for `#HYBRID_THRESHOLD` to something much higher with `M913` like so:

> M913 X Y Z E 200

I expect the noises will disappear.

It's a feature, but the limits in the standard configuration are set fairly low.

thisiskeithb, 2020-05-11:
I just put in https://github.com/MarlinFirmware/Configurations/pull/103 to disable `HYBRID_THRESHOLD` on the SKR Mini E3 configs.

ComputerKing12, 2020-05-11:
Thank you both a lot, this did fix my issue.

closed 2020-05-11 by ComputerKing12