MarlinFirmware/Marlin #17955: [BUG] Can't compile after PR #17953 (Fix host_action_notify calls passing SRAM pointers)

thisiskeithb, 2020-05-11:
# Bug Description

After PR #17953 "Fix host_action_notify calls passing SRAM pointers", I can no longer compile with `EMERGENCY_PARSER` enabled on `LPC1768`:

```prolog
In file included from Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/HAL.h:45,
                 from Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/HAL.h:26,
                 from Marlin/src/HAL/LPC1768/../../feature/../inc/MarlinConfig.h:30,
                 from Marlin/src/HAL/LPC1768/../../feature/host_actions.h:24,
                 from Marlin/src/HAL/LPC1768/../../feature/e_parser.h:31,
                 from Marlin/src/HAL/LPC1768/usb_serial.cpp:28:
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h:59:5: error: 'EmergencyParser' does not name a type
   59 |     EmergencyParser::State emergency_state;
      |     ^~~~~~~~~~~~~~~
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h: In constructor 'MarlinSerial::MarlinSerial(LPC_UART_TypeDef*)':
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h:47:9: error: class 'MarlinSerial' does not have any field named 'emergency_state'
   47 |       , emergency_state(EmergencyParser::State::EP_RESET)
      |         ^~~~~~~~~~~~~~~
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h:47:25: error: 'EmergencyParser' has not been declared
   47 |       , emergency_state(EmergencyParser::State::EP_RESET)
      |                         ^~~~~~~~~~~~~~~
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h: In member function 'virtual bool MarlinSerial::recv_callback(char)':
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h:55:7: error: 'emergency_parser' was not declared in this scope
   55 |       emergency_parser.update(emergency_state, c);
      |       ^~~~~~~~~~~~~~~~
Marlin/src/HAL/LPC1768/../../feature/../inc/../HAL/./LPC1768/MarlinSerial.h:55:31: error: 'emergency_state' was not declared in this scope
   55 |       emergency_parser.update(emergency_state, c);
      |                               ^~~~~~~~~~~~~~~
*** [.pio/build/LPC1768/src/src/HAL/LPC1768/usb_serial.cpp.o] Error 1
```

Dropping the commit (`00e7599`), I can compile successfully.

### My Configurations

[SKR 1.4-based config with the issue](https://github.com/thisiskeithb/Marlin/tree/myconfig/E3-SKR14-TMC2209) and [SKR 1.3-based config with the issue](https://github.com/thisiskeithb/Marlin/tree/myconfig/E3-SKR13-TMC5160)

### Steps to Reproduce

1. Use an SKR 1.4 or SKR 1.3 (`LPC1768` and probably other LPC-based boards), enable`EMERGENCY_PARSER`
2. Compile & observe errors

**Expected behavior:** Successful compile.

**Actual behavior:** Compilation fails

thisiskeithb, 2020-05-11:
This doesn't seem to affect `AVR` & of course `EMERGENCY_PARSER` isn't implemented on `STM32` yet, so I don't have anymore build types to test.

thisiskeithb, 2020-05-12:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/17953.

Evg33, 2020-05-12:
skr 1.4T LPC1769
enabled #define EMERGENCY_PARSER 
sources after https://github.com/MarlinFirmware/Marlin/commit/2f3077f2192981fa26659f990f37419c03c77159

```
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
In file included from d:\cod\marlin\marlin\src\hal\lpc1768\hal.h:45,
                 from d:\cod\marlin\marlin\src\hal\hal.h:26,
                 from d:\cod\marlin\marlin\src\inc\marlinconfig.h:30,
                 from d:\cod\marlin\marlin\src\feature\host_actions.h:24,
                 from d:\cod\marlin\marlin\src\feature\e_parser.h:31,
                 from Marlin\src\HAL\LPC1768\usb_serial.cpp:28:
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h:59:5: error: 'EmergencyParser' does not name a type
   59 |     EmergencyParser::State emergency_state;
      |     ^~~~~~~~~~~~~~~
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h: In constructor 'MarlinSerial::MarlinSerial(LPC_UART_TypeDef*)':
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h:47:9: error: class 'MarlinSerial' does not have any field named 'emergency_state'
   47 |       , emergency_state(EmergencyParser::State::EP_RESET)
      |         ^~~~~~~~~~~~~~~
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h:47:25: error: 'EmergencyParser' has not been declared
   47 |       , emergency_state(EmergencyParser::State::EP_RESET)
      |                         ^~~~~~~~~~~~~~~
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h: In member function 'virtual bool MarlinSerial::recv_callback(char)':
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h:55:7: error: 'emergency_parser' was not declared in this scope
   55 |       emergency_parser.update(emergency_state, c);
      |       ^~~~~~~~~~~~~~~~
d:\cod\marlin\marlin\src\hal\lpc1768\MarlinSerial.h:55:31: error: 'emergency_state' was not declared in this scope
   55 |       emergency_parser.update(emergency_state, c);
      |                               ^~~~~~~~~~~~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\usb_serial.cpp.o] Error 1
============================================= [FAILED] Took 12.38 seconds =============================================
```

thisiskeithb, 2020-05-12:
@Evg33 You have to wait for https://github.com/MarlinFirmware/Marlin/pull/17961 to be merged or pull in the changes from the PR to your branch.

thisiskeithb, 2020-05-12:
I'll reopen this issue for now so its clear that `EMERGENCY_PARSER` is still broken until that PR is merged.

commit 0fbedf5b (2020-05-12): Fix LPC host actions, add tests

Fixes #17955
closed 2020-05-12 by thinkyhead