MarlinFirmware/Marlin #17958: [BUG] (Copymaster 3D 300 DEFAULT files fail to verify & unknown board via USB)

Welsh-Dragon, 2020-05-11:
[image]

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step] Copy Copymaster 3d 300 files from examples to Marlin 2.0x
2. [Second Step] Click Tools > Get Board Info, See screenshot for info
3. [Third Step] Tools > Board > Arduino Mega or Mega 2560
4. [Fourth Step] Processor > ATmega2560 (Mega 2560)
5. [Fifth Step] Verify

**Expected behavior:** [What you expect to happen]
Default code to fully verify

**Actual behavior:** [What actually happens]
The following errors

Arduino: 1.8.12 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\marlinconfig.h:40:0,

                 from sketch\src\HAL\AVR\HAL.cpp:24:

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."

   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."

    ^~~~~

In file included from sketch\src\inc/MarlinConfig.h:40:0,

                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."

   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."

    ^~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:777:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1420:25: error: 'HOMING_BUMP_MM' was not declared in this scope

 constexpr float hbm[] = HOMING_BUMP_MM;

                         ^~~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1420:25: note: suggested alternative: 'X_HOME_BUMP_MM'

 constexpr float hbm[] = HOMING_BUMP_MM;

                         ^~~~~~~~~~~~~~

                         X_HOME_BUMP_MM

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1421:1: error: static assertion failed: HOMING_BUMP_MM requires X, Y, and Z elements.

 static_assert(COUNT(hbm) == XYZ, "HOMING_BUMP_MM requires X, Y, and Z elements.");

 ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1422:1: error: non-constant condition for static assertion

 static_assert(hbm[X_AXIS] >= 0, "HOMING_BUMP_MM.X must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1423:1: error: non-constant condition for static assertion

 static_assert(hbm[Y_AXIS] >= 0, "HOMING_BUMP_MM.Y must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1424:1: error: non-constant condition for static assertion

 static_assert(hbm[Z_AXIS] >= 0, "HOMING_BUMP_MM.Z must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

In file included from sketch\src\inc/MarlinConfig.h:40:0,

                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/SanityCheck.h:777:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^~~~~~~~~~~~~

sketch\src\inc/SanityCheck.h:1420:25: error: 'HOMING_BUMP_MM' was not declared in this scope

 constexpr float hbm[] = HOMING_BUMP_MM;

                         ^~~~~~~~~~~~~~

sketch\src\inc/SanityCheck.h:1420:25: note: suggested alternative: 'X_HOME_BUMP_MM'

 constexpr float hbm[] = HOMING_BUMP_MM;

                         ^~~~~~~~~~~~~~

                         X_HOME_BUMP_MM

sketch\src\inc/SanityCheck.h:1421:1: error: static assertion failed: HOMING_BUMP_MM requires X, Y, and Z elements.

 static_assert(COUNT(hbm) == XYZ, "HOMING_BUMP_MM requires X, Y, and Z elements.");

 ^~~~~~~~~~~~~

sketch\src\inc/SanityCheck.h:1422:1: error: non-constant condition for static assertion

 static_assert(hbm[X_AXIS] >= 0, "HOMING_BUMP_MM.X must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

sketch\src\inc/SanityCheck.h:1423:1: error: non-constant condition for static assertion

 static_assert(hbm[Y_AXIS] >= 0, "HOMING_BUMP_MM.Y must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

sketch\src\inc/SanityCheck.h:1424:1: error: non-constant condition for static assertion

 static_assert(hbm[Z_AXIS] >= 0, "HOMING_BUMP_MM.Z must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

In file included from sketch\src\MarlinCore.cpp:43:0:

sketch\src\module/motion.h: In function 'float home_bump_mm(AxisEnum)':

sketch\src\module/motion.h:126:51: error: 'HOMING_BUMP_MM' was not declared in this scope

   static const xyz_pos_t home_bump_mm_P PROGMEM = HOMING_BUMP_MM;

                                                   ^~~~~~~~~~~~~~

sketch\src\module/motion.h:126:51: note: suggested alternative: 'X_HOME_BUMP_MM'

   static const xyz_pos_t home_bump_mm_P PROGMEM = HOMING_BUMP_MM;

                                                   ^~~~~~~~~~~~~~

                                                   X_HOME_BUMP_MM

In file included from c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\marlinconfig.h:40:0,

                 from sketch\src\HAL\AVR\HAL_SPI.cpp:34:

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."

   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."

    ^~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:777:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1420:25: error: 'HOMING_BUMP_MM' was not declared in this scope

 constexpr float hbm[] = HOMING_BUMP_MM;

                         ^~~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1420:25: note: suggested alternative: 'X_HOME_BUMP_MM'

 constexpr float hbm[] = HOMING_BUMP_MM;

                         ^~~~~~~~~~~~~~

                         X_HOME_BUMP_MM

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1421:1: error: static assertion failed: HOMING_BUMP_MM requires X, Y, and Z elements.

 static_assert(COUNT(hbm) == XYZ, "HOMING_BUMP_MM requires X, Y, and Z elements.");

 ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1422:1: error: non-constant condition for static assertion

 static_assert(hbm[X_AXIS] >= 0, "HOMING_BUMP_MM.X must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1423:1: error: non-constant condition for static assertion

 static_assert(hbm[Y_AXIS] >= 0, "HOMING_BUMP_MM.Y must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:1424:1: error: non-constant condition for static assertion

 static_assert(hbm[Z_AXIS] >= 0, "HOMING_BUMP_MM.Z must be greater than or equal to 0.");

 ^~~~~~~~~~~~~

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

[300.zip](https://github.com/MarlinFirmware/Marlin/files/4609299/300.zip)

 
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-05-11:
Use this Configuration_adv.h https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Copymaster3D/300/Configuration_adv.h

ellensp, 2020-05-11:
vid.0=0x1A86 pid.0=0x7523
This is a CH340 usb/serial converter.   
You need to install windows drivers. See https://sparks.gogo.co.nz/ch340.html

Welsh-Dragon, 2020-05-11:
@ellensp thank you for the quick response, I have copied/pasted & saved the above & it still fails to verify...

Arduino: 1.8.12 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\inc/MarlinConfig.h:40:0,

                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/SanityCheck.h:777:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^~~~~~~~~~~~~

In file included from c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\marlinconfig.h:40:0,

                 from sketch\src\HAL\AVR\HAL.cpp:24:

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:777:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^~~~~~~~~~~~~

In file included from c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\marlinconfig.h:40:0,

                 from sketch\src\HAL\AVR\HAL_SPI.cpp:34:

c:\users\user\appdata\local\temp\arduino_build_371524\sketch\src\inc\SanityCheck.h:777:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.

     static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

     ^~~~~~~~~~~~~

exit status 1
Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ellensp, 2020-05-11:
Clearly no one has tried putting marlin on this type of machine in some time...
In Configuration.h you need to enable NOZZLE_PARK_FEATURE
In Configuration_adv.h enable ADVANCED_PAUSE_FEATURE 
This will get it to compile. 

NB.
BUT without access to this machine there is no way to know if any of the values related to these two features are correct. you will have to check them.

Welsh-Dragon, 2020-05-11:
I installed CH34x_Install_Windows_v3_4 closed & opened Marlin backup & it still says unknown board...

I will try to enable NOZZLE_PARK_FEATURE & ADVANCED_PAUSE_FEATURE
Upgrade files can be found on https://copymaster3d.com/pages/upgrades-downloads if this helps

The machine came out, late 2018 & has been added to Marlin 2.0.x between Jan - May 2020 as not seen at 4:04 minutes in 22 Jan 2020 in this video by [The First Layer] https://www.youtube.com/watch?v=-rOZpOiWYxM]

ellensp, 2020-05-11:
you need to unplug and replug in the printer board. this is a windows level issue.

Welsh-Dragon, 2020-05-11:
I have done that & had the machine switched on & off & still no avail, 

The ADVANCED_PAUSE_FEATURE is a synch to enable however the NOZZLE_PARK_FEATURE is indicating more than one Nozzle, see ` * Requires NOZZLE_PARK_FEATURE to park print head in case MMU unit fails.
 * Requires EXTRUDERS = 5
 *
 * For additional configuration see Configuration_adv.h
 */
//#define PRUSA_MMU2

// A dual extruder that uses a single stepper motor
//#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 0
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 0, 90 } // Angles for E0, E1[, E2, E3]
  #if EXTRUDERS > 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 1
  #endif
#endif

// A dual-nozzle that uses a servomotor to raise/lower one (or both) of the nozzles
//#define SWITCHING_NOZZLE
#if ENABLED(SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 0
  //#define SWITCHING_NOZZLE_E1_SERVO_NR 1          // If two servos are used, the index of the second
  #define SWITCHING_NOZZLE_SERVO_ANGLES { 0, 90 }   // Angles for E0, E1 (single servo) or lowered/raised (dual servo)
#endif

/**
 * Two separate X-carriages with extruders that connect to a moving part
 * via a solenoid docking mechanism. Requires SOL1_PIN and SOL2_PIN.`

Welsh-Dragon, 2020-05-11:
After waiting a very long time, it now verifies, saying 

> Sketch uses 139700 bytes (55%) of program storage space. Maximum is 253952 bytes.
Global variables use 4907 bytes (59%) of dynamic memory, leaving 3285 bytes for local variables. Maximum is 8192 bytes.
[Configuration_adv.zip]

Here are the updated, verified files, for ease of use...

(https://github.com/MarlinFirmware/Marlin/files/4610432/Configuration_adv.zip)

Welsh-Dragon, 2020-05-12:
Strange to me, **connecting thru USB via Pronterface**, using M115, **the Copymaster 3D 300 shows as Marlin FLASHFORGE**, yet **Marlin still says UNKNOWN BOARD,**  thereby as Pronterface is able to compile & show the knowledge on the board & Marlin cannot, then it is a Marlin issue & need to be debugged...

`Connecting...
start
Printer is now online.
echo:Marlin FLASHFORGE
echo: **Last Updated: 2017-10-10 12:00**| Author: (none, default config)
echo:**Compiled: Jun  6 2018**
echo: Free Memory: 3584  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X94.14 Y94.14 Z400.00 E96.27
echo:Maximum feedrates (units/s):
echo:  M203 X100.00 Y100.00 Z5.00 E20.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1000 Y500 Z100 E4000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R500.00 T500.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H210 B50 F255
M145 S1 H220 B60 F255
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:SD card ok
>>> m115
SENDING:M115
FIRMWARE_NAME:Marlin FLASHFORGE (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:EEPROM:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:SOFTWARE_POWER:1
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0`

What is interesting is Copymaster [came out late 2018 using 2017 firmware from **FLASHFORGE** ships April 2020 with outdated firmware compiled on the board in 2018, likely knowing the fire-risks of the **THERMAL-RUNAWAY** etc & provide **UNTESTED-not-working** programs on their website to UPGRADE-the-FIRMWARE**

Welsh-Dragon, 2020-05-13:
In boards.h Copymaster 3D is listed, & default says `#pragma once

#include "macros.h"

#define BOARD_UNKNOWN -1

//
// RAMPS 1.3 / 1.4 - ATmega1280, ATmega2560
//

#define BOARD_RAMPS_OLD               1000  // MEGA/RAMPS up to 1.2

#define BOARD_RAMPS_13_EFB            1010  // RAMPS 1.3 (Power outputs: Hotend, Fan, Bed)
#define BOARD_RAMPS_13_EEB            1011  // RAMPS 1.3 (Power outputs: Hotend0, Hotend1, Bed)
#define BOARD_RAMPS_13_EFF            1012  // RAMPS 1.3 (Power outputs: Hotend, Fan0, Fan1)
#define BOARD_RAMPS_13_EEF            1013  // RAMPS 1.3 (Power outputs: Hotend0, Hotend1, Fan)
#define BOARD_RAMPS_13_SF             1014  // RAMPS 1.3 (Power outputs: Spindle, Controller Fan)

#define BOARD_RAMPS_14_EFB            1020  // RAMPS 1.4 (Power outputs: Hotend, Fan, Bed)
#define BOARD_RAMPS_14_EEB            1021  // RAMPS 1.4 (Power outputs: Hotend0, Hotend1, Bed)
#define BOARD_RAMPS_14_EFF            1022  // RAMPS 1.4 (Power outputs: Hotend, Fan0, Fan1)
#define BOARD_RAMPS_14_EEF            1023  // RAMPS 1.4 (Power outputs: Hotend0, Hotend1, Fan)
#define BOARD_RAMPS_14_SF             1024  // RAMPS 1.4 (Power outputs: Spindle, Controller Fan)

#define BOARD_RAMPS_PLUS_EFB          1030  // RAMPS Plus 3DYMY (Power outputs: Hotend, Fan, Bed)
#define BOARD_RAMPS_PLUS_EEB          1031  // RAMPS Plus 3DYMY (Power outputs: Hotend0, Hotend1, Bed)
#define BOARD_RAMPS_PLUS_EFF          1032  // RAMPS Plus 3DYMY (Power outputs: Hotend, Fan0, Fan1)
#define BOARD_RAMPS_PLUS_EEF          1033  // RAMPS Plus 3DYMY (Power outputs: Hotend0, Hotend1, Fan)
#define BOARD_RAMPS_PLUS_SF           1034  // RAMPS Plus 3DYMY (Power outputs: Spindle, Controller Fan)
`

Line 56 // RAMPS Derivatives - ATmega1280, ATmega2560
Line 106 #define BOARD_COPYMASTER_3D           1150  // Copymaster 3D

[boards.h.zip](https://github.com/MarlinFirmware/Marlin/files/4619012/boards.h.zip)

boelle, 2020-06-20:
@Welsh-Dragon still an issue?

2357532, 2020-06-21:
@boelle
I didn't use Welsh-Dragon's config, but while trying to compile for the 400 version I encountered a similar error to his:

```
SanityCheck.h:752:5: error: static assertion failed: ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.
static_assert(nullptr == strstr(FILAMENT_RUNOUT_SCRIPT, "M600"), "ADVANCED_PAUSE_FEATURE is required to use M600 with FILAMENT_RUNOUT_SENSOR.");

```

Following @ellensp's advice:

```
In Configuration.h you need to enable NOZZLE_PARK_FEATURE
In Configuration_adv.h enable ADVANCED_PAUSE_FEATURE
```

led to a new error:

```
sketch\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope
 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

```

The cause of this was the use of "HOMING_BUMP_MM", which I believe may have been deprecated. The fix was to replace it with 

```
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2
```

copied from the current default config. It then compiled and uploaded. There was a Marlin error message about EEPROM settings, which were resolved with M502, M500.

After uploading, the default speeds of the x and z axes are much lower than factory settings. This seems strange but it's based on the default config, and I assume there's a good reason.

The only issue I (still) have, which is what led me here, is that the y-axis (bed) speed and/or acceleration seems to be too fast (in spite of conservative settings), which means that the motor makes a racket in spite of using some rubber to dampen it. This is probably due to the printer being fairly low budget. I couldn't find any information on the stepper motor in google.

The issue isn't closed, because I've only compiled for the 400 model and not forked. I may fork and add a config titled "Copymaster 3D 400 (updated June 2020)", but I'll need some time to make sure there are no major operating errors.

boelle, 2020-06-23:
@2357532 discussions should happen on the discord server

boelle, 2020-06-23:
i will close this one as @Welsh-Dragon have not reported back

closed 2020-06-23 by boelle