MarlinFirmware/Marlin #17959: [BUG] Double-click for babystep no longer works

Knifa, 2020-05-11:
### Bug Description

Double click for babystep no longer works on my Ender 3 Pro with SKR Mini E3 v1.2 on https://github.com/MarlinFirmware/Marlin/commit/1475fd312a1572e1d43978f669bce02a72f63dab.

Previous build was https://github.com/MarlinFirmware/Marlin/commit/6947dc139e4e81fed3986510eb32a93c58bd256c. I'll see if I can figure out where it broke.

I tried fiddling with the timer (doubled from `1250` which worked fine before to `2500`) but no dice.

The `Babystep Z` menu item still works.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4609931/config.zip)

### Steps to Reproduce

1. Enable `BABYSTEPPING` and `DOUBLECLICK_FOR_Z_BABYSTEPPING`.
2. Start a print.
3. When things getting moving, try double click.

**Expected behavior:** Babystep menu shows.

**Actual behavior:** Current menu item is selected instead (return to status).

Knifa, 2020-05-11:
52aa459 is the culprit.

Knifa, 2020-05-11:
This looks like the breaking change: https://github.com/MarlinFirmware/Marlin/commit/52aa459a2dc08a8f803ad9ffb7aa173dc449637f#diff-8685dde8286c4e15db3bf3e8f74d5e6fR227

Reverting back to `screen == status_screen` solves the problem.

I'm not sure if the other changes here maybe have similar effects.

swilkens, 2020-05-11:
@thinkyhead made this change

qwewer0, 2020-05-11:
Can confirm, but I have the same board.

Knifa, 2020-05-12:
Fixed by https://github.com/MarlinFirmware/Marlin/pull/17962.

closed 2020-05-12 by Knifa