MarlinFirmware/Marlin #17963: [BUG] EXPERIMENTAL_I2CBUS needs re-initialisation with Wire.begin(); (M261)

EsserPrototyping, 2020-05-11:
### Bug Description

When defining EXPERIMENTAL_I2CBUS, M261 should start an I2C request, but it is not. (No data is send via SDA or SCL).
It will work however if I add Wire.begin(); before every request. 

```
void GcodeSuite::M261() {
  if (parser.seen('A')) i2c.address(parser.value_byte());
  uint8_t bytes = parser.byteval('B', 1);

  Wire.begin(); // added

  if (i2c.addr && bytes && bytes <= TWIBUS_BUFFER_SIZE)
    i2c.relay(bytes);
  else
    SERIAL_ERROR_MSG("Bad i2c request");
}
```
So maybe Wire is not started, or it get´s stopped elsewhere I think.
When using I2C EEPROM, SDA and SCK are toggling fine, but M261 is also not doing anything.
With an older bugfix version I´ve used a few month ago, it was not working too until I2C EEPROM was enabled. (probably because it was starting Wire.. )

### Steps to Reproduce

1. #define EXPERIMENTAL_I2CBUS
2. send request via M261 (M261 A9 B4 in my case)
3. Nothing happens (no error warning)

#### Additional Information

Board: SKR PRO 1.1

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

robin-gdwl, 2020-06-29:
Having a similar problem: 
I call the request function from twibus.cpp and while the function is called no request is executed (the attached teensy board does not register a request. Sending works as expected.