MarlinFirmware/Marlin #17965: [BUG??] USB/nonATX PSU Dual Y race condition

sprintfailed, 2020-05-11:
### Bug Description

Fixed issue some time ago doesn't work as that guy's case has had ATX PSU

- Doing m502 all the time (even tried to disable EEPROM)
- All Stepper jumpers are removed
- Interchanging drivers/cables/steppers doesn't help
- Have tried bugfix and 2.0 master
- Have tried multiple suggestions here, no luck (but, being honest, those don't suite my issue :) )
- Have even tried multiple motherboards in configuration.h

Plug in USB, turn on PSU -> DUAL Y both steppers run smoothly, no noise, sun is shining
BUT
Turn on PSU first, plug in USB -> steppers lose their settings
AT THE SAME TIME
Plug in USB, turn on PSU, I click on-board "reset" button - steppers lose their settings as well

### My Configurations

MKS GEN L v2
Five TMC2209 (E0 is not connected to the Extruder as I'm building PCB CNC)
nonATX (12V/15A) PSU, is NOT PS_ON controllable
Identical Y steppers

### Steps to Reproduce

Plug in USB, turn on PSU -> DUAL Y both steppers run smoothly, no noise, sun is shining
BUT
Turn on PSU first, plug in USB -> steppers lose their settings, out of synchro
AT THE SAME TIME
Plug in USB, turn on PSU, I click on-board "reset" button - steppers lose their settings as well, out of synchro

**Reports on a good day:**
`Marlin bugfix-2.0.x
echo: Last Updated: 2020-05-09 | Author: (none, default config)
echo:Compiled: May 11 2020
echo: Free Memory: 4922  PlannerBufferBytes: 1152
echo:V78 stored settings retrieved (662 bytes; crc 3378)
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E80.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E300.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z100.00 E1500.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 I1 Y800
echo:  M906 T0 E800
echo:; Hybrid Threshold:
echo:  M913 X3 Y3 Z1
echo:  M913 I1 Y2
echo:  M913 T0 E3
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Y
echo:  M569 S1 T0 E
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Y2 connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW`

Please note, ALL LOW - because PSU hasn't been engaged yet. I engage PSU and m122:

`SENDING:M122
		X	Y	Y2	Z	E
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	1436	1436	1436	1436	1436
MAX current	2025	2025	2025	2025	2025
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual	31/31	31/31	31/31	31/31	31/31
PWM scale	36	36	36	36	36
vsense		0=.325	0=.325	0=.325	0=.325	0=.325
stealthChop	true	true	true	true	true
msteps		8	8	8	8	8
tstep		max	max	max	max	max
PWM thresh.	52708	52708	65535	65535	52708
[mm/s]		-	-	-	-	-
OT prewarn	false	false	false	false	false
off time	3	3	3	3	3
blank time	36	36	36	36	36
hysteresis
 -end		-3	-3	-3	-3	-3
 -start		6	6	6	6	6
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Y2	Z	E
sg_result	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:1F:00:00
		Y	0xC0:1F:00:00
		Y2	0xC0:1F:00:00
		Z	0xC0:1F:00:00
		E	0xC0:1F:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Y2 connection... OK
Testing Z connection... OK
Testing E connection... OK`

**Reports on my regular day:**

`Marlin bugfix-2.0.x
echo: Last Updated: 2020-05-09 | Author: (none, default config)
echo:Compiled: May 11 2020
echo: Free Memory: 4922  PlannerBufferBytes: 1152
echo:V78 stored settings retrieved (662 bytes; crc 3378)
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E80.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E300.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z100.00 E1500.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 I1 Y800
echo:  M906 T0 E800
echo:; Hybrid Threshold:
echo:  M913 X3 Y3 Z0
echo:  M913 I1 Y0
echo:  M913 T0 E3
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Y
echo:  M569 S1 T0 E
Testing X connection... OK
Testing Y connection... OK
Testing Y2 connection... OK
Testing Z connection... OK
Testing E connection... OK
Printer is now online.`

`>>> M122
SENDING:M122
		X	Y	Y2	Z	E
Address		0	0	0	0	0
Enabled		false	true	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual	12/31	12/31	12/31	12/31	12/31
PWM scale	14	38	14	14	14
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		16	16	16	16	16
tstep		max	max	max	max	max
PWM thresh.	3294	3294	**65535**	**65535**	**3294**
[mm/s]		3	3	**-**	**-**	3
OT prewarn	false	false	false	false	false
off time	3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
 -end		-1	-1	-1	-1	-1
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Y2	Z	E
sg_result	0	0	0	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:0C:00:00
		Y	0xC0:0C:00:00
		Y2	0xC0:0C:00:00
		Z	0xC0:0C:00:00
		E	0xC0:0C:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Y2 connection... OK
Testing Z connection... OK
Testing E connection... OK`

You see? Basically, race condition :(

#### And some frequently questioned settings, so I don't spend your time:

//#define PSU_CONTROL
//#define SQUARE_WAVE_STEPPING

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4611556/Marlin.zip)

**Units/sec:**
//#define DISTINCT_E_FACTORS
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 80 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 300 }
//#define LIMITED_MAX_FR_EDITING
#define MAX_FEEDRATE_EDIT_VALUES    { 600, 600, 10, 600 } 
#define DEFAULT_MAX_ACCELERATION      { 1500, 1500, 100, 1500 }

**5 Drivers**
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
#define Y2_DRIVER_TYPE TMC2209
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209
//#define E1_DRIVER_TYPE TMC2209

**Microsteps** (no changes, actually to any, apart from Y2 pins definition)
 #if AXIS_IS_TMC(Y)
    #define Y_CURRENT       800
    #define Y_CURRENT_HOME  Y_CURRENT
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
  #endif

  #if AXIS_IS_TMC(Y2)
    #define Y2_CURRENT      800
    #define Y2_CURRENT_HOME Y2_CURRENT
    #define Y2_MICROSTEPS    Y_MICROSTEPS
    #define Y2_RSENSE         Y_RSENSE
    #define Y2_CHAIN_POS     Y_CHAIN_POS
    #define Y2_SERIAL_TX_PIN   44
    #define Y2_SERIAL_RX_PIN   66
  #endif

**THRESHOLDS**
  #define X_HYBRID_THRESHOLD     100
  #define Y_HYBRID_THRESHOLD     100
  #define Y2_HYBRID_THRESHOLD    100
  #define Z_HYBRID_THRESHOLD       3
  #define E0_HYBRID_THRESHOLD     100

boelle, 2020-06-20:
@sprintfailed is this still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sprintfailed, 2020-07-01:
Sorry for being inactive. I haven't resolved the problem :( This fact made me try GRBL out. I've lost some features, yes, but at least machine works alright now :)

Cheers.

thinkyhead, 2020-07-06:
All I know is that Trinamic steppers must have 12V / 24V power or they cannot be configured over SPI / Serial. So in Marlin, if the power is disabled and later re-enabled, the `Power::power_on` method calls `restore_stepper_drivers` (e.g., in response to `M80` or with `AUTO_POWER_CONTROL` enabled). If power conditions are such that stepper drivers lose their settings, and the firmware isn't aware of the occurrence, it can't restore the lost settings.

What do you propose that we change in order to make your startup procedure work?

ManuelMcLure, 2020-07-06:
@thinkyhead I've been pondering either adding an option to ``M122`` or adding a new gcode to reinitialize the TMC drivers (i.e. do the same thing that ``M80`` does).
Currently, to initialize the drivers if you don't have ``PSU_CONTROL`` enabled you need to use ``M502`` or ``M501`` which will clobber any changes you might have done to values not yet stored in EEPROM. Or you must have ``PSU_CONTROL`` enabled which will let you use ``M80`` (consecutive ``M80`` will reinitialize the drivers each time even if there is no ``M81`` between them).
Maybe ``M122 I`` could be used to initialize the drivers?

thinkyhead, 2020-07-07:
> Maybe M122 I could be used to initialize the drivers?

That would be a good addition

But, it seems like the steppers should be properly set up if the board is reset while there is PSU power. Are there issues in that kind of situation? And if so, does any amount of `PSU_POWERUP_DELAY` help?

ManuelMcLure, 2020-07-07:
If the board is reset with the PSU already powered up the drivers should already be properly initialized. If that's not happening for the OP there's some other issue at play. I'll take a look at the configuration files and see if there's anything that stands out.

ManuelMcLure, 2020-07-07:
Also, perhaps this is part of the problem that was introduced in the TMCStepper 0.7.0 library and is fixed in 0.7.1.

closed 2020-08-11 by github-actions[bot]