MarlinFirmware/Marlin #17966: [BUG] Latest Bugfix2.0 - Y Stutter from Hell

thierryzoller, 2020-05-11:
- JD Deactivated
- Build from today bugfix2.0

What's wrong ?
Just look at this video :
https://youtu.be/buVWpHUtP0o

Note: The click sound you here is my broken fan - that's expected.

> >>> m503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E444.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z50.00 E40.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X2000.00 Y2000.00 Z100.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P800.00 R10000.00 T2000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.30 E5.00

> 
Marlin bugfix-2.0.x
echo: Last Updated: 2020-05-11 | Author: (ZOLLER, BLTouch)
echo:Compiled: May 11 2020
echo: Free Memory: 4737  PlannerBufferBytes: 1312
echo:V79 stored settings retrieved (826 bytes; crc 33782)

thisiskeithb, 2020-05-11:
Please follow the Issue Template and attach your configs.

thierryzoller, 2020-05-11:
Honestly, I am getting a bit tired of pulling latest bug-fix0.20 to debug one issue and getting another one. I feel like threading water ;)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4612178/Marlin.zip)

Here are the configurations, that haven't changed.

kmobs, 2020-05-11:
Is it when you're doing curves? May be due to junction deviation? Try disabling  #define JD_USE_LOOKUP_TABLE in planner.h 

https://github.com/MarlinFirmware/Marlin/issues/17920

thierryzoller, 2020-05-11:
@kmobs  : Stutter happens as soon as Y moves, on the straight line, on the first diagonal line (you see and hear it) it's worse when printing a circular shape (like in the video)

thierryzoller, 2020-05-11:
JD is disabled

GhostlyCrowd, 2020-05-11:
> Is it when you're doing curves? May be due to junction deviation? Try disabling #define JD_USE_LOOKUP_TABLE in planner.h
> 
> #17920

He doesn't have JD enabled, his configs defines Classic Jerk. Going all the way back to 2.0.2 is going to be a stretch for anyone to narrow down when this was introduced on your platform.  Try 2.0.5.3 if it still occurs roll back to the previous release and so on.

GhostlyCrowd, 2020-05-11:
Also, a heads up it's not just your Y that's missing steps, the X is clearly doing it as well, and if i had to take a guess your Extruder as well. Z moves way too slow for you to see it do it but id wadge its doing it as well.  

You're running TMC 2100's  correct?

thierryzoller, 2020-05-11:
Not sure, it is an Artillery X1 - I found this. 
https://fulament.com/products/artillery-sidewinder-x1-original-oem-stepper-drivers

Indeed X axis aswell, I edited my comment above to reflect that.

GhostlyCrowd, 2020-05-11:
I think you're suffering from the classic TMC2100's from MKS not having their CGF1 grounded which leaves them in stealthchop which is a known culprit for missing steps badly on TMC2100.  The fact you said it did it always (not always as bad) makes me think this further more. 

Did you get the 2100's from mks as well?  

[image]

MKS missed soldering this pin to Ground CFG1 to set the mode properly for Spreadcycle which is definitely what you want on 2100, especially on a bed slinger.

GhostlyCrowd, 2020-05-11:
Oh i just saw your link you have the knockoff AT2100 stepper drivers, i'd probably get rid of those. Thats always been your problem.

thierryzoller, 2020-05-11:
I am not sure what to make of this, how would downgrading to 1.9 or 2.0.5 fix the issue? 
Here is a close up of the driver : https://youtu.be/733xwFRjoOE?t=441

When I said it "always" did it, I meant that it stuttered with JD enabled when printing circles, which we know isn't related to my driver. Believe me when I say that "it did it always" means, in particular situations, I had near perfect prints, so I'd say that it is unlikely hardware related. That said considering he sales volume of artillery, many many thousand would be affected then.

thierryzoller, 2020-05-11:
@GhostlyCrowd : I may upgrade the drivers in the future - but I don't think that is the issue I am seing? Or has anything been introduced that would particularly impact these drivers?

GhostlyCrowd, 2020-05-11:
Basically, your drivers are an Undocumented knock off of the TMC2100, and they are piggy backing on the TMC libraries, as marlin moves forward and merges in new features and the TMC library is updated, your AT2100 stepper drivers are choking on it. Original TMC2100's wouldn't suffer from these issues. 

Basically, there is no information on the AT2100 it's a straight up Chinese clone with no documentation and there is no support in marlin for them technically, and its impossible to trouble shoot for the reasons above.

thierryzoller, 2020-05-11:
That makes a lot of sense. So for all standard artillery users it would be Game over after a certain marlin version then. I will check the VREF tomorrow, if that's not the issue, I will swap them for Originals - any recommendation?

GhostlyCrowd, 2020-05-11:
> That makes a lot of sense. So for all standard artillery users it would be Game over after a certain marlin version then. I will check the VREF tomorrow, if that's not the issue, I will swap them for Originals - any recommendation?

the quickest and easiest drop in and go you are already configured hardware and software wise, for it would be to get some TMC2100's

thisiskeithb, 2020-05-11:
The [stock Sidewinder X1 config](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Artillery/Sidewinder%20X1) is configured for 2100s, but it's easy enough to update if you move to new drivers.

thinkyhead, 2020-05-11:
We recently changed the default value for `MINIMUM_STEPPER_POST_DIR_DELAY` from 20 to 60. As one troubleshooting step, try defining this in your configuration with a value of 20.

Of course, as always, don't forget to send `M502`, `M500` to reset the EEPROM.

GhostlyCrowd, 2020-05-11:
> We recently changed the default value for `MINIMUM_STEPPER_POST_DIR_DELAY` from 20 to 60. As one troubleshooting step, try defining this in your configuration with a value of 20.
> 
> Of course, as always, don't forget to send `M502`, `M500` to reset the EEPROM.

Beat me to the pop i was just browsing recent merges to see if it was something simple to try.

thierryzoller, 2020-05-12:
MINIMUM_STEPPER_POST_DIR_DELAY is commented out in both working and non working versions (configuration_adv.h)

randellhodges, 2020-05-12:
If you are using the latest bugfix, then commenting it out in configuration_adv.h means it is going to use the default, which changed to 60.

He was explicitly asking that you do not comment it out, and set it to 20 to force the "old" value of 20.

thierryzoller, 2020-05-12:
Got it - thanks will try.

markusressel, 2020-05-18:
I am having the same issue on an Artillery Genius.
You can see the firmware I use here:
https://github.com/markusressel/Marlin/tree/artillery-genius

I have already tried to disable `S_CURVE_ACCELERATION` because I had it enabled alongside `LIN_ADVANCE` and it seems like it did change the sound of the jittering a bit, but the effects are still the same.
I normally use Cura to slice my models but tried PrusaSlicer for a quick test - **with the same bad results**.

As my fork is more than a couple days behind upstream (this repo) I am almost certain changing `MINIMUM_STEPPER_POST_DIR_DELAY` won't change it, since I built my version of the firmware way before that.

I have **not** enabled `CLASSIC_JERK` so `JUNCTION_DEVIATION_MM` should be in effect.

In my tests I mostly used 50mm/s as a test speed, printing directly on the printer using the the USB stick, without an USB cable attached to the printer so Octoprint doesn't interfere in any way. Although printing at slower speeds limits the effect of the stuttering it only hides the problem and even at 20mm/s it is still very much noticable in the finished print.

Here are some pictures:
[image]
This is the underside of a circular test print ([this one](https://www.thingiverse.com/thing:11895)).  
The left was sliced in Cura, `S_CURVE_ACCELERATION` still turned on.  
The right one was sliced in PrusaSlicer, `S_CURVE_ACCELERATION` turned off. since PrusaSlicer doesn't have the option to print circular infill the center is lines, but it is still visible on the outer ring.

[image]
This is the top of the prints, left still Cura, right Prusa.

[image]
This is a closeup side shot of the Cura print. You can clearly see a wobbly pattern, which, if printing a pipe or something, would go all the way through it.

[image]
This is the Prusa print and it already looks a lot better on the outside, but the stuttery motion is still present and visible from the top view (second image).

I am not sure if this is driver related, since to me it looks like the printer is also slightly retracting filament when these jitters occur. Thats why I initially thought it might be related to Cura.

thierryzoller, 2020-05-18:
Have you tried with classical jerk enabled?

markusressel, 2020-05-18:
No, as I assumed - judging from the previous conversation in this thread - that it does not help.

thierryzoller, 2020-05-18:
Give it a try nonetheless please

markusressel, 2020-05-18:
Ok so I updated my branch to the latest upstream and disabled Junction Deviation with [this change](https://github.com/markusressel/Marlin/commit/45773b1fbff918406094b706b22ebe538adf6f66)
and what can I say, the stuttering is **completely gone** on the gcode sliced with Cura. There are some weird "infill" moves between the lines of the circles that still juggle around, but I think that is completely unrelated to the circle stuttering, which is not happening anymore.
I messed up the print quite a bit because the firmware update changed my bed leveling settings all over the place, but I will attach a picture when the print is complete.

thierryzoller, 2020-05-19:
This confirms me suspicion, I seem to have provided the Configs from another build instead of the ones I compiled it with. Will redo the tests with JD deactivated on bug-fix2.0

thierryzoller, 2020-05-19:
I broke my heatbreak, ordered a new one. Will update after I fixed my hardware.

thierryzoller, 2020-05-19:
@markusressel : Your third picture looks like your extruder or LA went crazy ? Tha'ts not supposed to be normal is it?

markusressel, 2020-05-19:
* the firmware flashing messed up my bed leveling and I fiddled around with the height while the print was going
* I use gluestick for bed adhesion so my bed was pretty messy
* I might not have the best cura settings for the first layer yet, since my printer is only about a week old and I didn't invest the time to figure this out yet. The rest of the print looks fine though.

boelle, 2020-06-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-06-20 by boelle

thierryzoller, 2020-06-21:
@boelle - @thinkyhead  set the title of this bug report. 
It's not closed as its not clear/not resolved.

thierryzoller, 2020-06-21:
This comment remains true:
I broke my heatbreak, ordered a new one. Will update after I fixed my hardware.

I received the remainder of the parts yesterday only.

boelle, 2020-06-21:
it was a question right? the title says [Q]

DardoMX, 2020-08-10:
[AT2100_Datasheet_CN_V0.1.pdf](https://github.com/MarlinFirmware/Marlin/files/5052957/AT2100_Datasheet_CN_V0.1.pdf)

0-mcclock-0, 2020-09-20:
Not sure if you're still troubleshooting this, but I was also having stuttering on my Sidewinder X1 after upgrading from stock firmware to the custom G2 firmware based on Marlin 2.0.X bugfix. Coincidentally, I had previously seen CNC Kitchen's Youtube video about stuttering caused by the mainboard not being able to keep up with the volume of gcode commands. I suspected that with LA and JD enabled on the new firmware, the mainboard might not have been able to keep up anymore.

I did what he suggested and increased the values of "Maximum Resolution", "Maximum Travel Resolution", and "Maximum Deviation" in the "Mesh Fixes" section of Cura. Apparently they are set unnecessarily low by default. I doubled my values, that seems to have fixed my stuttering issue.