MarlinFirmware/Marlin #17968: [BUG] ATmega2560 FAST_PWM_FAN duty cycle limited to 50%

g3gg0, 2020-05-11:
Hello everyone,
on my GT2560 board, after migrating to marlin v2.0, the PWM fan only goes up to 50%.
I have a custom laser module on my printer, like shown [here](http://www.g3gg0.de/wordpress/uncategorized/3d-custom-geeetech-i3-printhead-and-some-other-stuff/) and thus could not
control it anymore using the PWM duty cycle.
Measured the signal with a scope and the M106 Sx for x=[0..255] maps to [0..50]% duty cycle.

Made a quick fix for me, modifying Marlin\src\HAL\AVR\fast_pwm.cpp and double the value passed to _SET_OCRnQ().
This is of course just a crude hack, making things work again.

I am not working on head, but on cdbc7cc52c9bc0f1889384e8db2dc46a93f3fb94 which doesnt have too much changed since then.
Hardware is a Geeetech GT2560 (BOARD_GT2560_REV_A) and FAST_PWM_FAN is enabled, but no custom frequency.

Can someone else confirm this bug?

Regards,
g3gg0

thisiskeithb, 2020-05-11:
Update to the latest `bugfix-2.0.x` and see if your issue is resolved.

g3gg0, 2020-05-11:
sorry, cannot test right now.
from checking the changed files since then, i cannot see anything related would have changed.

when going through the source, i can find a potential condition that would cause this:
in fast_pwm.cpp:277
>     _SET_OCRnQ(timer.OCRnQ, timer.q, 2 * v * float(top / v_size)); // Scale 8/16-bit v to top value`
the intention was most likely to scale the given duty cycle (based on 255 as maximum) to the capcom top value.
but the float cast happens *after* the integer division already happened, thus only is correct in the case top==v_size.
this assumes top *is always* 255.

Conditionals_post.h:2135
>   #define FAST_PWM_FAN_FREQUENCY ((F_CPU) / (2 * 255 * 1)) // Fan frequency default

gives us a freq of 31372 Hz. 
(hm why not 31250 by using even numbers? but i guess this is some backwards compatibility thing)
i would be surprised when this would in any situation result in the timer top value being 255.

so if you fix the mentioned line to:
>     _SET_OCRnQ(timer.OCRnQ, timer.q, v * float(top) / float(v_size)); // Scale 8/16-bit v to top value

this issue should be resolved and the duty cycle values suddenly would make sense.

Regards,
g3gg0

g3gg0, 2020-05-12:
can you confirm this bug?

thisiskeithb, 2020-05-12:
> Update to the latest `bugfix-2.0.x` and see if your issue is resolved.

g3gg0, 2020-05-13:
i updated to head and this line is still buggy.

g3gg0, 2020-05-16:
Hello. 
The bug was present in the commit i posted above, it was present in the branchs head 4 days ago and i pointed out the buggy line.
Is there anything else I can do?

Regards

sjasonsmith, 2020-05-16:
If you already know how to fix it, you are free to submit a pull request with your change.

If you want anybody else to look at it, you will need to attach your configuration files as requested in the issue template.

thinkyhead, 2020-05-25:
Thanks for the patch. That's a sticky one, and as we are deep into the laser code lately it makes perfect sense. Ordinarily the `map` macro would be used, and maybe it still can be….

closed 2020-05-25 by thinkyhead

g3gg0, 2020-05-26:
you are welcome

commit f776260d (2020-05-25): Fix OCR / PWM calculation (#18094)

Fixes #17968