MarlinFirmware/Marlin #17969: Commented DEACTIVATE_SERVOS_AFTER_MOVE servo still deactivates

c3D-Dan, 2020-05-11:
### Bug Description

Deactivate servo after move DEACTIVATE_SERVOS_AFTER_MOVE has been commented out, servo is deactivated after approximately 10 seconds.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4612645/Marlin.zip)

### Steps to Reproduce

1. Comment out DEACTIVATE_SERVOS_AFTER_MOVE in configuration.h
2. T0 to move to servo position 1
3. Wait 10 seconds.  Servo becomes deactivated.  Same with T1

**Expected behavior:** Servo should remain active
**Actual behavior:** Servo Deactivates

thinkyhead, 2020-05-12:
- Do you see the same issue with another board?
- Have you ruled out a hardware issue?

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

c3D-Dan, 2020-05-12:
Sorry guys.  Jumped the gun.  Issue was with servo.  Doesn't occur with another servo.

closed 2020-05-12 by c3D-Dan