MarlinFirmware/Marlin #17971: [FR] IMPROVEMENT: Harmoize platformio env names with motherboard declaration for easier beginner configuration

bjoernwedi, 2020-05-12:
Currently there is no link in between the platformio.ini and the motherboard declaration in the configuration.h **except for MAYLAN 200**.

For the `#define MOTHERBOARD =` there is already a `#ifndef` wrapper, which would empower in platform.ini already a `build_flags = ... -DMOTHERBOAD=`.

I would recommend to define this by: `build_flags = -DMOTHERBOARD=$PIO_ENV` and to rename the environments accordingly to match ENV NAME and BOARD NAME.

Secondly it would be helpful to be able to define SERIAL_PORT and SERIAL_PORT2 (and maybe BAUD_RATE) also from platformio.ini ENVs by wrapping them in configuration.h also with `#ifndef`.

Especially for unexperienced users, this would reduce the number of build errors due to the fact of not aliging BOTH files simultaniously.

bjoernwedi, 2020-05-12:
If I would get a recommendation of structure, I would like to participate on this.
BTW: this harmonization can be done with a plenty of things defined in platformio.ini, like the HAL paths, ... .
If we would apply patterns in between CPUs, platforms, ... and the folder structure, comfiguration becomes more convenient by CRITERIA <=> PATH PATTERN and less maual/hard coded configuration.

thisiskeithb, 2022-04-09:
This is not possible for all environments / boards since some environments cover many boards.

You will now get an error if you’ve selected the incorrect environment and will be presented with a list of correct envs to compile with.

closed 2022-04-09 by thisiskeithb