MarlinFirmware/Marlin #17973: [FR] Teensy 4.1 board support in Marlin

Hedda, 2020-05-12:
Teensy 4.1 was just released today and compared to Teensy 4.0 (see #16666 feature request by @limitz ) the biggest hardware news with Teensy 4.1 is a MicroSD-card slot, more I/O support, and integrated 10/100 Mbit Ethernet PHY (physical layer) support

https://www.cnx-software.com/2020/05/12/teensy-4-1-board-gets-longer-for-ethernet-microsd-slot-and-more-gpios/

About Teensy 4.1's new features:
https://pjrc.com/teensy-4-1-released/

Teensy 4.1 product page:
https://pjrc.com/store/teensy41.html

Teensy 4.1 Arduino compatible board is powered by the same NXP i.MX RT1062 Cortex-M7 crossover processor clocked at 600 MHz as Teensy 4.0, however Teensy 4.1 is about doubling in length in order to add a 10/100 Mbit Ethernet PHY, a MicroSD card slot, and offer more I/Os.

Teensy 4.1 specifications:

- SoC – NXP i.MX RT1062 Arm Cortex-M7processor at 600 MHz with  1024KB RAM (512KB is tightly coupled),
- Storage – 8 MB serial flash (64KB reserved for recovery & EEPROM emulation), MicroSD Socket, footprints for 2x extra QSPI chips such as flash or 8MB PSRAM chip
- USB – 1x micro USB port for power and programming
- Networking – 6-pin Ethernet header via 10/100 Mbit DP83825 PHY
- Expansion via through-holes and pads
  - USB – 2x USB ports, both 480 MBit/sec (5 pins with power management)
  - Storage – 1x SDIO (4 bit) native SD
  - Audio – 2x I2S Digital Audio, 1x S/PDIF Digital Audio
  - Serial – 8x Serial, all with 4 byte FIFO; 3x CAN Bus (1x with CAN FD)
  - 35x PWM pins
  - 40x digital pins, all interrupt capable
  - 18x analog pins, 2x ADCs on chip
  - 3x SPI, all with 16 word FIFO; 3x I2C, all with 4 byte FIFO
  - Total I/O pins – 55 (vs 40 for Teensy 4.0)
- Security – Cryptographic Acceleration; Random Number Generator
- Misc – RTC for date/time, 32x general-purpose DMA channels, Programmable FlexIO, Pixel Processing Pipeline, Peripheral cross triggering, Power On/Off management
- Power Supply – 5V via micro USB port
- Dimensions – 7.2 x 1.8 cm (TBC)

The software is mostly software compatible with Teensy 4.0, and can also be programmed in Arduino IDE using Teensyduino software add-on.

EvilGremlin, 2020-05-12:
Don't forget to send hardware to interested developers ;)

Hedda, 2020-05-12:
> Don't forget to send hardware to interested developers ;)

I'm sure Marlin developers can contact PJRD for donation, (I have no affiliation to them or any others).

Hedda, 2020-05-12:
FYI, as per #16666 @limitz was working on a branch for Teensy 4.0 here:

https://github.com/limitz/Marlin/tree/teensy40

He, however, put that project on the backburner:

>  I was just breadboarding it with some external mosfets and tmc5160s, don't think you need much more than that. However, I got lost in some preprocessor hell because several dependencies of marlin don't use the marlin HAL (obviously) and the arduino port for teensy didn't work for them (u8g2 for instance). Also tried to fix some PROGMEM stuff instead of just defining it as NOP, but that just opens another can of worms with bugs introduced by the fact that a lot of HALs "fix" pgm compilation errors by just defining it empty (serial writes expecting progmem string called with strings allocated on heap or stack.
> In the meantime I got an ultratronic board and put this project on the backburner, because I think that the way marlin is structured now is flawed and would rather spend time on fixing those flaws than working around those flaws to get the teensy working :)

EvilGremlin, 2020-05-12:
I don't think it works like "hey some random guys want us to do complex unpaid work so we can support your new board, please send a few to us for free"... Well, first and foremost developer must be personally interested.

Hedda, 2020-05-12:
> I don't think it works like "hey some random guys want us to do complex unpaid work so we can support your new board, please send a few to us for free"... Well, first and foremost developer must be personally interested.

@EvilGremlin I know that is not how open source development projects works, this is only a feature request, as in a suggestion, however, Teensy 4.1 does look like it would be the perfect match for Marlin so my guess is that it will only be a matter of time before there are developers with a personal interest.

SJ-Innovation, 2020-05-12:
I have a few on the way, and would love to see & help this happen.

CrazzyFrenchDude, 2020-08-13:
Hello guys, so happy and excited to finally be on Github, as a fast learner I have been snicking here and there in the 3D community, reading silently and learning. All this began with the need of a Pick and Place Machin for my "daily" DIY's!Then found out that the makers were using 3D printing controller boards for that matter! And some would have issues with using the thermistors input to plug in there vacuum sensor or other little annoying problems which are not real because the 3D controllers were not especially built for that (Pick And Place Machines)! So I went to the Open Pnp Forum and launched a thread, asking them what would their dream controller for OPNP be. Then started a humble design pcb design around the new comer Teensy 4.1, the PeeNaPle_V1.1b. With it's modularity it joins 3 worlds, not so different! 3D printing, CNC and Pick and Place Machines! With an array of extension boards, stepper driver adapters, Mosfet switches, extension boards, etc and not to mention what the makers will come up with!
Still in revision mode I just can't wait for Marlin to be supported or GrblHAL or any gerber firmware! So I give you an early glance at this little baby!
[image]

rarens, 2020-09-03:
@Hedda I have created a fork to support Teensy 4.1.  So far it compiles and I somehow got the SD card working.  I'm a mobile dev by trade and I really don't have a clue what I'm doing. All of the nondescript defines in mcu programming make my head spin. I just tinkered with it until it worked.   I would really like to get the SD to show up as a mass storage device.  So far I've been unsuccessful.  Any help would be greatly appreciated.  I would ultimately like to submit a pull request so that teensy 4.0/4.1 are officially supported. 

I did my best on the implementation of the timer HAL. I'm pretty sure it works but I really don't know how to verify it.  

https://github.com/rarens/Marlin/tree/teensy41

CrazzyFrenchDude, 2020-09-03:
Check this fork https://github.com/bilsef/Marlin/tree/Teensy4.x

Gamester17, 2020-10-09:
Looks like Teensy 4.1 (and Teensy 4.0) support has now been merged via https://github.com/MarlinFirmware/Marlin/pull/19311 via PR from @bilsef

closed 2020-10-09 by ellensp

Gamester17, 2020-10-09:
@rarens did you test / verify the pull request from @bilsef in https://github.com/MarlinFirmware/Marlin/pull/19311 pull request?

nanotuxi, 2020-10-09:
@Gamester17 
tested on a bare teensy41 board. It's working connected to RepetierHost with dry-run enabled.
See [my comment](https://github.com/MarlinFirmware/Marlin/pull/19311#issuecomment-702851080).

Gamester17, 2020-10-15:
Another FYI, just read about newly announced "iMX RT1064 uCOM" board from Embedded Artists so submitted a feature request:

https://github.com/MarlinFirmware/Marlin/issues/19749

Maybe depending on its price when general availability it might be an option for those looking at developing BoBs for Teensy 4.1