MarlinFirmware/Marlin #17974: [BUG] (short description)

ruggb, 2020-05-12:
### Bug Description

When I compile the new version of Marlin-bugfix-2.0.x.219812e I get this error multiple times, maybe for all the .CPP files.

Error while detecting libraries included by C:\Users..............

The compile does finish = no fatal errors

It does not occur on a previous build. 

### My Configurations
Compiled with default config files, straight out of the box.

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. unzip download
2. dbl click Marlin.ino
3. click Verify

**Expected behavior:** verify build without errors

**Actual behavior:** multiple lines for maybe all .CPP files
Error while detecting libraries included by C:\Users..............

The compile does finish = no fatal errors

It does not occur on previous builds.

#### Additional Information
Windows 10 pro 1909 18363.815
Arduino 1.8.12

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-05-13:
What was the include, and in which file?

ruggb, 2020-05-13:
false alarm. I have too many builds open and got confused.

closed 2020-05-13 by thinkyhead

thinkyhead, 2020-05-13:
Please use the following resources before coming to GitHub, and please only post bugs that you are unable to solve for more than …let's say…  a week:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

ruggb, 2020-05-13:
this isn't solved. If I remove some files I get a lot less errors, but I still get errors.
I do not get any errors compiling a previous build b7f8163 with exactly the same tools/system and default config.
Why is that?
There is something Arduino doesn't like about the libraries referenced in the build. As best I can tell, all the files referenced in the errors include MarlinConfig.h but I don't think it is in that file. Probably one that it includes.

thinkyhead, 2020-05-13:
It's recommended to use PlatformIO if Arduino IDE is giving trouble. To most efficiently build with Arduino you need to delete all the HALs from the project that you're not using. If building on Windows, don't put the project into a location with a long path name.

And, please use the provided support forums and get your help with building Marlin from the user community. We don't want to have conversations here unless it is with other developers working on solutions to bugs, or in direct investigation of issues.

If you do want to report a bug, use fewer words and provide data in a concise form. There is nothing actionable here, so this issue is closed.