MarlinFirmware/Marlin #17975: Can't compile without Z_SAFE_HOMING

ruggb, 2020-05-12:
### Bug Description

I have a nozzle probe/Z_endstop. Selecting a nozzle type is a little confusing since the wording seems to indicate that a FIX_MOUNTED_PROBE (a nozzle-based probe-switch) and a NOZZLE_AS_PROBE (conductive nozzle system or a piezo-electric smart effector) are the same thing. Mine is a piezo that is activated by the nozzle touching the bed. Previously, FIX_MOUNTED_PROBE was a nozzle probe. Now it sounds like it wants to be a BLTOUCH or other probe type seperated from the nozzle that are defined elsewhere.

Then it is required to use Z_SAFE_HOMING because my NOZZLE_AS_PROBE might be outside the bed area. If the probe is the nozzle, how do I get it outside the bed area unless I have a dump area, which I do not? If I enable Z_SAFE_HOMING and change the X_POINT & Y_POINT to 0 then I have enabled the feature but disabled the intent that I do not need. So why do you make it a requirement? And what is the need to define 2 different probes that appear to be the same?

Then this error message is also confusing:
"Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue." I got the error because it is not enabled ==> commented out. So it is telling me to do what I have already done. But still, I will get an error for something I don't need.

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4616488/Marlin.zip)

### Steps to Reproduce

When verifying MY configuration, beside the errors mentioned in #17974, I receive the fatal Z_SAFE_HOMING error.

**Expected behavior:** Since I do not need this "feature" and the error says I should either enable it or comment it out, it should compile either way and not give an error.

**Actual behavior:**Won't compile, fatal error.
#### Additional Information

Marlin-bugfix-2.0.x.219812e

ellensp, 2020-05-12:
read the error 
#error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue.""
You have to make a decision. either enable Z_SAFE_HOMING or disable the check in SanityCheck.h.
You don't not need Z_SAFE_HOMING

ruggb, 2020-05-12:
How do you read this line:
"Enable it or comment out this line to continue."
as
"disable the check in SanityCheck.h."?
Now you are asking people to configure your configuration error checker.
Does that seem as absurd to you as it does to me?
That doesn't resolve the issue that I do not need Z_SAFE_HOMING and the only way to get around it is to enable it and set X&Y to 0 thus disabling it. Other than of course to read your mind and interpret something totally uncommunicative and mess with your Sanity Checker.
You can do better than that!

trouch, 2020-05-12:
@ruggb your X/Y min could be outside of the bed and defaulting to (0,0) might be not relevant if you haven't set X/Y_MIN_POS properly, and Marlin can't guess that.
So using Z_SAFE_HOMING and setting X/Y location for Z homing seems appropriate.

ruggb, 2020-05-12:
Z_SAFE_HOMING as detailed in the comments indicates that it is to prevent a nozzle mounted probe from homing off the bed. It probably would be better stated as a carriage mounted probe. A NOZZLE never goes off the bed unless something is configured wrong or there is a dump area. My probe IS the nozzle, not something mounted on the carriage. I don't have a dump area, therefore, my nozzle never will go off the bed. So, I do not require Z_SAFE_HOMING.
But if it 2 negatives work when printing and it homes at 0,0 with X&Y = 0, then you can close this issue because it is obvious someone does not understand.

ManuelMcLure, 2020-05-12:
Many common printers have their end stops set such that the nozzle is off the bed when they home.

ruggb, 2020-05-12:
So how do you program Marlin so all the others that do not don't have to wrestle the code down to make it work for them?
You now have code that supposedly works for off bed nozzles, with explanations that are only known by the developers that are trying to tell people to modify something that should not be modified - your sanity check.

txt4nk, 2020-05-13:
I also cannot compile my firmware because of this. When compiling it WITH Z safe homing enabled, my BLtouch does not work.

ruggb, 2020-05-13:
@ManuelMcLure 
> 
> 
> Many common printers have their end stops set such that the nozzle is off the bed when they home.

Can you tell me which one do? I would like to look at them as I am curious as to why they would do that. Thanks

sjasonsmith, 2020-05-13:
I’m not really sure why this is even being debated. Printers do it. Even without a good reason that would be true.

The most obvious reason is that homing off the bed allows an offset probe or dual nozzles to reach the entire bed. 

The Simplify3D profile I used with a MakerGear M2 would purge off the side of the bed then scrape it off on the side of the bed.

ManuelMcLure, 2020-05-13:
> @ManuelMcLure
> 
> > Many common printers have their end stops set such that the nozzle is off the bed when they home.
> 
> Can you tell me which one do? I would like to look at them as I am curious as to why they would do that. Thanks

The Anet A8 for one - the endstops are at X=-33 Y=-15 or so. It's just a matter of how they're put together. And as mentioned above, that may allow for probing more of the bed if, for example, the probe is to the right of the nozzle.

Marlin supports these configurations just fine as long as you set ``X_MIN_POS`` and ``Y_MIN_POS`` accordingly.

thinkyhead, 2020-05-13:
Several printers, when homed, have the nozzle in a position which is off of the bed. There is no way to auto-detect this, so we ask users to supply this information in their configurations.

Back in the Marlin pre-1.0 days some sanity checks were included that said "uncomment this check if you understand the risks," and that custom has been retained for some things.

closed 2020-05-13 by thinkyhead