MarlinFirmware/Marlin #17978: [BUG] "no media" message when plugging in an SD card

leytanelectronics2, 2020-05-12:
I am having a problem with my version of marlin not seeing the SD card when I plug it in and when I do it shows the "media inserted" text so it can tell there's a card there but when I go into the menu and go to where the usual print from media text should be it just shows "no media" even though there is a card in it.

I have made no modifications to the programming outside of activating the filament runout sensor, adding the scrolling text for file names, and showing the "show remaining time" function in the config-adv.h file.

The motherboard I am using is a ramps board using an Arduino mega 2560 as the processor, if anybody has found a fix for this please post it here with clear instructions, my skills at programming are practically nonexistent so I need all the help I can get.

https://www.youtube.com/watch?v=-UbTaDt3aiE this is a video made by someone having the same issue I am having, but my set up is using the LCD 12864 Graphic Smart Display instead of the one shown in this video.

P-C-R, 2020-05-12:
please post you config.h and config_adv.h in a zip file here

P-C-R, 2020-05-12:
Connect your printer via USB to your PC and test this command

https://marlinfw.org/docs/gcode/M021.html

Is the SD Card formated in FAT32 ?

leytanelectronics2, 2020-05-12:
the default was just FAT so I believe it is set to that.

leytanelectronics2, 2020-05-12:
ok, i tried it, didn't work, tried fat32, worked for a second then could no longer see the data on the card.

thinkyhead, 2020-05-13:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

leytanelectronics2, 2020-05-13:
@thinkyhead thank you, I'll test it as soon as I can.

leytanelectronics2, 2020-05-13:
@thinkyhead  ok i fixed the problem, i changed the spi-speed setting from half to a quarter and now it works perfectly.

closed 2020-05-13 by leytanelectronics2