MarlinFirmware/Marlin #17981: linear advance slow prntng

Crunch69, 2020-05-12:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

XDA-Bam, 2020-05-12:
It's not clear to me, what "very slow" means in this context. Can you quantify this? Also, looking at your configs, there's a couple of potential problems:

- Your default extruder jerk of 5 is too low for LA in my experience. I run 15 on an Ender 3 Pro, which should have the same extruder. If the extruder jerk is too low, LA will reduce printing speeds to respect it.
- You are running Junction Deviation with a value of 0.02. If your print is mostly slow in corners and curves: That's why. Try 0.04, maybe even up to 0.08, depending on how stiff your frame is and how much ringing you are willing to accept.
- You are using S-Curve Acceleration with LA. That is not recommended and officially not supported, as far as I know. Deactivate it and check again.

If these config changes do not fix it, please let us know at which release it was still running fine for you (with the exact same config).

margaale, 2020-05-13:
Sorry to jump in, but why is  scurve accelereatiion incompatible with LA? I've been using these both settings for a long time now, just asking thou.

EvilGremlin, 2020-05-13:
LA don't account for SCurve correction, so result is quite unpredictable.

rado79, 2020-05-13:
> LA don't account for SCurve correction, so result is quite unpredictable.

maybe needs sanitycheck ?

qwewer0, 2020-05-13:
> maybe needs sanitycheck ?

It has.

rado79, 2020-05-13:
i have/had both enabled, never saw a warning - or do i misunderstand sanity check?

daleckystepan, 2020-05-13:
Check is not in stable version yet, only bugfix-2.0.x

rado79, 2020-05-13:
ok, but i do bugfix-2.0.x update once a week. no sanity check warnings in VSCode.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4624621/Configuration.zip)

qwewer0, 2020-05-14:
> ok, but i do bugfix-2.0.x update once a week. no sanity check warnings in VSCode.
> 
> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4624621/Configuration.zip)

This is what you should see if you try to compile it with both LA & S-C enabled and `EXPERIMENTAL_SCURVE` is disabled:
`#error "LIN_ADVANCE and S_CURVE_ACCELERATION may not play well together! Enable EXPERIMENTAL_SCURVE to continue."`

rado79, 2020-05-14:
@qwewer0 yes, disabling `EXPERIMENTAL_SCURVE` shows the warning. thanks for enlightening me.

Crunch69, 2020-05-16:
XDA-Bam thank you deactivated S-Curve Acceleration changed default extruder jerk to 15

closed 2020-05-16 by Crunch69