MarlinFirmware/Marlin #17982: After firmware upload, no boot, blank LCD and unresponsive printer

Trail3lazR, 2020-05-12:
Hi there!

Just recently I flashed the latest Marlin bugfix version to my Ender 3 Pro, I have the 1.1.5 silent board therefore the bootloader was already installed on it.

I chose the melzi_optiboot environment to compile the firmware and since then my printer is bricked.

It does not boot up most of the time (LCD backlit, but blank) and when it does it immediately freezes when trying to connect it to the pc.

Pronterface hangs on connection and trying to reflash fails with either:

"COM6 access denied"

"avrdude: stk500_getsync(): not in sync: resp=0xff"

depending on environment. (Melzi or Melzi_optiboot)

Any help would be much appreciated.

XDA-Bam, 2020-05-12:
How big is your sketch? Larger than 99% of the flash?

Trail3lazR, 2020-05-12:
I think it was a bit bigger than that, around 99.3% if I recall correctly.

XDA-Bam, 2020-05-12:
Try to deactivate a couple of features and flash again. You might need to reflash the bootloader, not sure. I had a very similar problem after flashing an image with >99% size on my stock Ender 3 Pro (1.1.4 board, no bootloader). Nothing helped afterwards, not even flashing the stock Creality image using Progisp. But after compiling the default Creality firmware which is only around ~90% size-wise, everything was back to normal.

ellensp, 2020-05-13:
From the product page https://www.creality3dofficial.com/products/creality-silent-mainboard-v1-1-5 "Bootloader installed (ISP/USB Arduino)"
It is it not a optiboot but a standard arduino bootloader.

Trail3lazR, 2020-05-13:
Allright, thanks. My suspicion was as well that the environment I chose was wrong. Well I have to wait for the ISP programmer to arrive in the mail :(

ellensp, 2020-05-13:
Any arduino or raspbury pi can be used as a ISP programmer...   as can a old parallel port and a few resistors if your desperate...

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle