MarlinFirmware/Marlin #17986: [BUG] M42 does not write PWM to pin

ThorsAvenger, 2020-05-13:
### Bug Description

M42 does not write write a PWM value to the pin, only S0 and S255 produce results in the form of LOW and HIGH.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. M42 P11 S128 
2. M42 P11 S0
3. M42 P11 S255

**Expected behavior:** [What you expect to happen]
1. Pin11 has DC of 50%
2. Pin11 has DC of 0%
3. Pin11 has DC of 100%

**Actual behavior:** [What actually happens]
1. Nothing happens
2. Pin11 has DC of 0%
3. Pin11 has DC of 100%

#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4621816/Marlin.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-20:
@ThorsAvenger still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle