MarlinFirmware/Marlin #17987: SWITCHING_EXTRUDER not changing Servo angles

sergiofagundes, 2020-05-13:
### Bug Description

I'm using SINGLENOZZLE / SWITCHING_EXTRUDER with a 2 extruders.

I can move the servo angles with M280, but if I use T0 or T1 the servo only moves to angle 0, not the ones I defined with #define SWITCHING_EXTRUDER_SERVO_ANGLES { 135, 50 }.

### My Configurations

Follow my configurations anexed.

### Steps to Reproduce

1. M280 P0 S50 ; work, servo moves
2. T0 ; doesnt work, servo moves to 0
3. T1 ; doesnt work, servo moves to 0

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4621876/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4621877/Configuration_adv.h.txt)

sergiofagundes, 2020-05-14:
Also I've tested the stepper motor with T0/T1, and the motor is not inverting direction between tool change.

sergiofagundes, 2020-05-14:
I tried firmware 2.0.5.3 and the Switching extruder works perfectly.

So this is a bug in bugfix-2.0.x.

c3D-Dan, 2020-05-28:
I have the same problem.  I've found that injecting 
"T1;
T0;
T1;"
At the start of a print causes the switching to work properly thereon.

boelle, 2020-06-21:
@sergiofagundes still an issue?

c3D-Dan, 2020-06-21:
Yes

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle