MarlinFirmware/Marlin #17991: [BUG] Z2 not stepping after X movement

aaronbutton, 2020-05-13:
### Bug Description
I am using a SKR 1.3 with TMC2209 drivers. I managed a poor print and on subsuquent tries the x axis stopped moving. 
I have both Z and Z1 opperating but after moving the Z axis Z1 wont enable automatically after an 0. 

### My Configurations

BTT SKR 1.3 mainboard
TMC2209 Driver
BL Touch
Sensorless homing

### Steps to Reproduce

1. Start printer
2. Move Z axis, both motors drive correctly
3. Move X axis, motor drives correctly
4. Move Z axis, only Z motor drives
5. Move Y axis, motor drives correctly
6. Disable steppers
7. Move Z axis, both motors drive
8. Move X axis, motor doesnt move.
9. Disable steppers
10. Move X, motor drives correctly
11. Move Y, motor drives correctly
12. Move Z, sometimes both motors drive correctly, 
13. Move X, motor doesnt move.
 
**Expected behavior:** All steppers to drive when a jog command is sent

**Actual behavior:** See steps to reproduce

#### Additional Information

I have tried recompiling and intialising EEPROM but the problem persists. Disabled linear advance.
**Attachments**
I have included my config, config advanced and Pins_BTT_SKR_1_3
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4623131/Marlin.zip)

boelle, 2020-06-20:
@aaronbutton still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle