MarlinFirmware/Marlin #17992: Help implementing 8 button keypad shift register (similar to reprapworld keypad)

jnour91, 2020-05-14:
Hi Team, 

I have spent too much time on this trying to figure it out and now I need you desperate help!!!

I have a 8 button keypad, that I have written code to get a 8 bit binary code for each button pressed coming from one data line (it works the same as reprapworld keypad)

I am now trying to implement this into marlin, 

I have started by creating a .h file of my code, and #include into ultra.cpp 

I have gotten an error staying I havent declared my pins (I am declaring them with port manipulation eg  "const int latchPin = (1<<PD6)" but its not recognizing it, I cant use a IDE pin as atmega 2560 doesnt use the pin i need "Logical pin 49 and 48" 

Your help will be highly appreciated, 
If someone is willing to spend time helping me with this via a instant msging or meeting app I will be happy to pay for your service 

Thank you

ellensp, 2020-05-14:
Duplicate of 17539, please close one of them.

ellensp, 2020-05-14:
You mean physical pins 48 and 49, commonly set up as

	PD	, // PD 5 ** 84 ** D84
	PD	, // PD 6 ** 85 ** D85

see https://forum.arduino.cc/index.php?topic=310792.msg2156093#msg2156093 to see how to add your own board with all mega pins...

jnour91, 2020-05-26:
Thanks ellensp, 

So below is my code for the keypad to run, 
I am still unsure how to call up the specific pins, in the current below code I have used "PD5_PIN" however this still doesn't work, the difficulty is that before implemting i was able to serial.print to see i am getting a binary output, however now I am assuming my implementation in ultralcd for the keypad to work is correct, and that the issue is how i have called the pins.   

Its been a while but I am back on it, 
`#include "fastio.h"
#include "fastio_1280.h"
//Initialize pins
 
 #define clockPin  PD5_PIN
 #define latchPin  PD6_PIN
 #define dataPin   PD7_PIN //int as it is a read only pin
//const int clockPin = (48); 
//const int latchPin = (49);
//int dataPin = (50); //int as it is a read only pin
//#define clockPin 1<<PD5;

void CREATBOT_SHIFTREG() {
   // Serial.begin (9600); //serial monitor
  SET_OUTPUT (clockPin); //outputs 
  SET_OUTPUT (latchPin); //outputs 
   SET_INPUT (dataPin); //inputs

// pulse latchpin (TO STORE THE DATA IN THE 74HC165)
WRITE(latchPin, LOW);   //LOW stores D0- D7 in IC
delayMicroseconds (20);
WRITE(latchPin, HIGH);   //HIGH stores DS in IC 

//Begin shiftin (send out data from IC read state 8 times)
int i; //initialization variable 
byte pinstate; //pin state to show if button is pressed 
byte buttons_creatbot_keypad = 0; //

// when clock is set high (rising edge) data pin will output from ic and input into arduino PD6 either 0 or 1 
// in this section pinstate will show 8 bits on println 
//when a button is pressed (ie data into pd6 when clock is high = 0) the pinstate will = 0 (example button 2(d1) is pressed pin state should show 0b11111101
 for ( i = 7  ; i >= 0; i--)  //The register transmits the information about the pins from pin 7 to pin 0 hence countdown 
    {
        
        WRITE(clockPin, LOW); //set clock pin low 
        delayMicroseconds(0.2);
        WRITE(clockPin, HIGH); //set clock pin high to send data to datapin
  
        if (READ(dataPin)) //digital read data pin if its high (ie 1) 

            {
            pinstate = 1;   //If dataPin = High then pinstate = 1
            buttons_creatbot_keypad = buttons_creatbot_keypad | (1<<i);
            
           } else {
             pinstate =0; //otherwise pin state equals zero 
            }
          }  

int j; //initialization variable 
byte pinstate2; //pin state to show if button 5 is pressed 

for ( j = 8  ; j >= 0; j--)  // 

    {
     
        WRITE(clockPin, LOW); //set clock pin low 
        delayMicroseconds(0.2);
        WRITE(clockPin, HIGH); //set clock pin high to send data to datapin
       
        if (READ(dataPin)) //digital read data pin if its high (ie 1) 
            {
            pinstate2 |= 1;   //If dataPin = High then pinstate = 1
       
           } else {
             pinstate2 =0; //otherwise pin state equals zero 
            }
           pinstate2  <<= 1;  //shift the bits along?
          }           
}`

boelle, 2020-06-21:
@jnour91 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle