MarlinFirmware/Marlin #17995: [FR] Remote extruder kinematics

diglooo, 2020-05-14:
### Description
What if we have the filament drive gear remotely driven by a timing belt, driven from a motor located on the X carriage? This should achieve no-backlash, direct driven and lightweight extruders.

This "remote extruder" (RE) motor will have to compensate the X motor movement in a differential fashion.
[image]

### Feature Workflow
Users who adopt this feature will have to specify the details of the kinematic chain in the config file.
- Ratio between pulley radius at X motor and RE motor shafts (needed to compute the RE motor shaft motion to achieve zero filament motion)
- Usual steps/mm for the Extruder motor, which is now "virtual".

I have the skills and commitment to build the hardware, compile and test the software but I have no knowledge of the internal Marlin structure and working principle to autonomously implement this feature.

Cheers by diglo from Italy.

EvilGremlin, 2020-05-14:
Ouch... I can feel the pains of poor planner... And hunders, hundreds of hours spent making it work... How do you plan on integrating it with LA and S-Curve?

diglooo, 2020-05-14:
Shouldn't the RE motor motion just be X+E if the motors are mounted like the attached picture and E-X if the X and RE motor are coaxial? Sorry but i have no clue about the inner workings of Marlin kinematics computaitions... :(

EvilGremlin, 2020-05-14:
Without corrections - yes, in theory it's no different from CoreXY. In reality? At high speeds? I suspect extrusion will suffer after 70-80mm/s even without LA and Scurve.
Though your idea doesn't make sense anyway. If you put it onto XY extruder Z bed - it'll jsat make one axis twice heavier, thus it's a massive downgrade. If you but it on XZ extruder Y bed - bed is still as heavy as motor, thus it's just useless.

diglooo, 2020-05-14:
For cartesian gantry style printers, like prusa, ender 3 and many others,
the XZ carriage is already holding the weight of the extruder motor, the
advantage of this solution is having a direct drive extruder without having
the X carriage carrying the extruder motor around...

EvilGremlin, 2020-05-14:
Yep, but it doesn't make bed any lighter, so this is no better than pancake with gearbox or belt

diglooo, 2020-05-14:
It does reduce the moving mass though.

EvilGremlin, 2020-05-14:
What's the point reducing mass of only one axis? Your speeds and quality only as good as heaviest carriage.

diglooo, 2020-05-14:
The point is to get at least one axis better. 
The bed stays at a constant height while printing; the X carriage is lifting upwards while printing and as a result you end up with a mass of ~200grams being accelerated back and forth at say 2000mm/s^2 at an height of 300mm from the bed surface if you're printing a tall model. The lighter the moving parts, the better... or not?

EvilGremlin, 2020-05-14:
Yep, to an extent. Remember that while your light X carriage go up your bed become heavier and heavier... IMO, it's better to build proper rigid and heavy CoreXY and don't waste time on half measures.