MarlinFirmware/Marlin #17997: [FR] (feature request title)

Rubenazzzo, 2020-05-14:
Hi folks.
I do not know if it is the best place to request this function, not only for me, but also in general.

The function consists, in people who have more than one unified extruder such as cyclops, when you have a power cut or a program stop, reset or sudden action that forces you to stop printing, when restarting the machine know that "T .." It has been used last time, (it should be written in eeprom) and an automatic option to reset the filament position can be executed, after a drastic action, and in turn issue a screen message that the extruder is without decoding, and even prevent it from a new impression is started without having done the priming.
Currently it occurs to me to use the option of custom commands and add the filament priming functions, manually for each head, but there is always the possibility of activating the opposite head and decocking more, which in my case with direct extruder, would the gear filament.

Example of actions on T0 and extruder 30mm:

[image]

1.heat the hot end to set temperature
[image]
2. wait a few seconds programmed
[image]
3.pull filament the programmed length
[image]
4 cool extruder
[image]
5 reset head and ready the printer in standbye
[image]

sorry if it's not place or way

Roxy-3D, 2020-05-14:
> It has been used last time, (it should be written in eeprom) 

This is a non-starter...   The EEPROM has a limited number of erase and rewrite cycles.  You will burn up the EEPROM pretty quickly doing a write to the EEPROM every time you switch tools.

Rubenazzzo, 2020-05-14:
Ok,well, other solución, is posible, Marlin can write on a sd o USB (method of run a print) a temporaly file, with a simple configuración , can read this options ,last tool,... Etc.
The same function a power_loss,

EvilGremlin, 2020-05-14:
I'm sure power loss recovery will get some love if BTT's micro UPS boards will succeed.