MarlinFirmware/Marlin #17998: [BUG] TMC 2208 connection error on Creality V2.2.1 Silent Mainboard

savenco, 2020-05-15:
Hi,

When compiling marlin for this board, the TMC drivers are soldered onto this board, so not a possibility to install a UART pin (i would think this would already be soldered into the board as with the SKR E3 mini V1.2)
Now these TMC 2208 are only working if you put them into _Standalone modus, they are using the same pins to communicate as the stock drivers on creality boards do.

I cannot get marlin to communicate with these drivers now... i'm pretty sure it has something to do with the PIN definitions but i cannot find the board included in the marlin 'boards' file?

Thanks for helping me out! :)

[configurationH.zip](https://github.com/MarlinFirmware/Marlin/files/4632581/configurationH.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

oliver-eifler, 2020-05-15:
like the creality silent boards 1.1.4/1.1.5 the tmc drivers on the silent board 2.2.1 are hardwired as standalone.

boelle, 2020-06-20:
@savenco still an issue?

savenco, 2020-06-21:
> @savenco still an issue?

Yes,
Works only in standalone but it should be manageable no?

boelle, 2020-06-21:
> like the creality silent boards 1.1.4/1.1.5 the tmc drivers on the silent board 2.2.1 are hardwired as standalone.

so not unless you make changes to the board

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle