MarlinFirmware/Marlin #18001: [BUG] error: 'strcmp_P' was not declared in this scope; on activate EXPECTED_PRINTER_CHECK

drackp2m, 2020-05-15:
For the past few days I'm trying to activate and test Marlin's weirdest and most hidden features (which my modest printer allows me to), and in this case, it's been impossible for me to figure out what the hell may be going on.

### Bug Description
The bug is that when I try to activate the `EXPECTED_PRINTER_CHECK` option, which currently in the latest version of the default branch can be found in [this line](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration_adv.h#L2764).

(Although whenever I make any changes to my configuration, I pull and merge my branch with the latest bugfix-2.0.x branch)

The result of the compiation is the following error:
```
Marlin/src/gcode/host/M16.cpp: In static member function 'static void GcodeSuite::M16()':
Marlin/src/gcode/host/M16.cpp:35:7: error: 'strcmp_P' was not declared in this scope; did you mean 'strcmp'?
   35 |   if (strcmp_P(parser.string_arg, PSTR(MACHINE_NAME)))
      |       ^~~~~~~~
      |       strcmp
*** [.pio/build/LPC1769/src/src/gcode/host/M16.cpp.o] Error 1
```

### My Configurations
[Configuration_2.0.5.2_15-05-20.zip](https://github.com/MarlinFirmware/Marlin/files/4634774/Configuration_2.0.5.2_15-05-20.zip)

### Steps to Reproduce
1. Download source code.
2. Apply my personal settings.
3. Try to build.

#### Additional Information
I also add the file pins_BTT_SKR_V1_4 in which I modified the pins for Z, E0 and E1. Although I don't think this should affect the mentioned error.
[pins_BTT_SKR_V1_4.h.zip](https://github.com/MarlinFirmware/Marlin/files/4634777/pins_BTT_SKR_V1_4.h.zip)

Thank you for your time, and I am very sorry if I have put the stop and this is not a bug, and I have simply encountered a problem whose origin I do not understand, and whose solution I cannot find.

ellensp, 2020-05-15:
You have found a standard arduino function that is not part of the lpc1768/9 arduino implementation. 
If you add the following line It will get past it. 
`#define strcmp_P(a, b) strcmp((a), (b))`
But its not a final fix...

drackp2m, 2020-05-16:
> You have found a standard arduino function that is not part of the lpc1768/9 arduino implementation.
> If you add the following line It will get past it.
> `#define strcmp_P(a, b) strcmp((a), (b))`
> But its not a final fix...

And where should I place this line? I already looked before your answer, that there is a file that contains this function. In fact it is the only one in the entire Marlin project.

[Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_macros.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_macros.h)

And I also noted that the UHS_macros.h file is only included in this one:

[Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_host.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_host.h)

And in turn the latter, UHS_host.h appears as a warning / error in several files in the [Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host) directory and imported file in the file [Marlin/src/sd/usb_flashdrive/Sd2Card_FlashDrive.cpp](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/sd/usb_flashdrive/Sd2Card_FlashDrive.cpp)

But from what you just told me, it seems that the error is a matter of the developers of the board.

drackp2m, 2020-05-16:
Okay, I just added the function you told me, just below the definition of the steppers in my board's pin configuration file (SKR 1.4) and it has compiled beautifully.

Thank you very much, I leave the issue open in case there is a more elegant way to correct this problem, but I, although I have been programming for many years, I do not have much idea of C, so I do not want to propose any nonsense.

ellensp, 2020-05-26:
has been added to bugfix, please close issue

closed 2020-05-26 by drackp2m