MarlinFirmware/Marlin #18002: [BUG] BLTouch 3 Probing failed (but behaves ok)

Haschtl, 2020-05-15:
### Bug Description

I'm facing this issue for some time now and it has not been fixed. 
I am probing at the beginning of each print with G30 and I then save the current z-offset.

Everything looks fine: The buildplate comes near to the probe, the probe deployes, the bed moves towards the probe slowly, the probe gets triggered and the buildplate lowers again.

But then the print aborts. In the terminal, i can see a "Probing failed" error - even if everything seems ok.

This could also be a misconfiguration issue from my side (even if it worked with my configurations ~3 month ago) but then, the error message could contain more information about the real issue.

### My Configurations

Board: SKR Pro v1.1
Marlin-branch: bugfix-v2.0.x
My fork (up to date): [click me](https://github.com/Haschtl/Marlin/tree/kellerdrucker)
### Steps to Reproduce

1. Run G30
2. Wait until probing finished
3. Read the "Probing failed" error message

**Expected behavior:** [What you expect to happen]

No "Probing failed" error

**Actual behavior:** [What actually happens]

Probing seems ok, but reports an error

GhostlyCrowd, 2020-05-15:
You need to ```#define USE_PROBE_FOR_Z_HOMING``` if you're using the probe as your z home and not a limit switch. 

Since you're  not defining ```Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN``` and using the probe pin instead.

DroneMang, 2020-05-15:
I have an SKR PRO and just did the above changes and it works for me on the newest bugfix.

Haschtl, 2020-05-16:
Thanks for your fast reply! 

I dont exactly understand... i dont want to use the probe for homing.
I have a z-MAX limit switch for homing 
And the probe (not as z-min limit switch) at z-min ( The setup is like an ultimaker)

Do i need this define anyway?

chiragkrishna, 2020-05-16:
had the same issue,... 
1) make sure your probe is not probing outside the endstop position(most common problem)
2) make sure your z-min position permits it, make it something like -5 or -10

GhostlyCrowd, 2020-05-17:
>make sure your z-min position permits it, make it something like -5 or -10

If your using an end-stop for your homing then the above is most likely your issue, you shouldn't have to make the defines i mentioned earlier for your setup. 

If this doesn't help, we would have to enable probe debugging and get some logs to investigate what the probe is doing.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle