MarlinFirmware/Marlin #18007: [BUG] Issue with HOME_BEFORE_FILAMENT_CHANGE

qwewer0, 2020-05-16:
### Bug Description

When `HOME_BEFORE_FILAMENT_CHANGE` is enabled, running the [Test.zip](https://github.com/MarlinFirmware/Marlin/files/4638263/Test.zip) gcode from SD card, it causes the printer to go into "Print Paused" and to home the printer, then any command after it will be executed just fine. At the end of the gcode I get M600: Too Cold message on the LCD.

Test.g contents:
```
M502						; Factory Reset
M92 X80 Y80 Z400 E137.7				; Set Axis Steps-per-unit
M201 X3000.00 Y3000.00 Z100.00 E10000.00	; Print Max Acceleration
M203 X200.00 Y200.00 Z20.00 E60.00		; Max Feedrate
M204 P500.00 R1000.00 T750.00			; Starting Acceleration
M205 J0.08 E15					; Advanced Settings
```
Test.g file is a snippet from my settings backup gcode. I narrowed it down to those commands in it. Removing any of those commands will cause the problem not to happen.

### My Configurations

SKR Mini E3 v1.2, Ender 3
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4638270/Configuration.zip)

thisiskeithb, 2020-05-16:
Is your hotend up to temp/above `EXTRUDE_MINTEMP` before issuing a filament change?

qwewer0, 2020-05-16:
Do I issuing a filament change? But no, the hotend is at room temperature.

With `HOME_BEFORE_FILAMENT_CHANGE` disabled this issue doesn't happens, only "M600: Too Cold" message on the LCD after the end of the file, but not always.

thisiskeithb, 2020-05-16:
> Am I issuing a filament change? But no, the hotend is at room temperature.

[`M600`](https://marlinfw.org/docs/gcode/M600.html) is the command for filament change, so your hotend will need to be up temp. That’s why you’re getting the “too cold” message.

qwewer0, 2020-05-16:
Even if there is no M600 in the Test.g file?

thisiskeithb, 2020-05-16:
If you’re not issuing `M600`, then `HOME_BEFORE_FILAMENT_CHANGE` shouldn’t come into play.

> With HOME_BEFORE_FILAMENT_CHANGE disabled this issue doesn't happens, only "M600: Too Cold" message on the LCD after the end of the file, but not always.

Can you clarify “but not always”? Is this with the same exact gcode printed over and over again? It sounds intermittent like your SD card is going bad and corrupting gcode/commands if results are not consistent.

Try using a different SD card and reslicing to see if you can consistently cause the problem to occur.

qwewer0, 2020-05-16:
It always or never happens with a specific firmware. (Minor changes in between)
But (if we can) let's ignore the "M600: Too Cold" message part, as it happened before this issue, but never caused any problem.

The main thing is that the printer 
> go into "Print Paused" and to home the printer

when using the Test.g file.

The SD card should be fine, as the file on it is (bit by bit) the same as it is in my PC.

thisiskeithb, 2020-05-16:
The issue is intermittent, so you need to narrow down how and what causes it (without changing variables) so it’s repeatable for proper troubleshooting.

qwewer0, 2020-05-16:
When I enable `HOME_BEFORE_FILAMENT_CHANGE` it (go into "Print Paused" and to home the printer) happens, and when I disable `HOME_BEFORE_FILAMENT_CHANGE` it doesn't happens. I recompiled, reflashed and re tested 5 times, with the same test gcode and got the same results.

Let's forgot about the M600 message at the end, as for that I need to dig deeper into my firmware settings.

thisiskeithb, 2020-05-16:
Can you post the full gcode you're trying to print instead of the test snippet?

Also, update to the latest `bugfix-2.0.x` to see if your issue persists.

qwewer0, 2020-05-16:
The full gcode: [Auto0.zip](https://github.com/MarlinFirmware/Marlin/files/4638445/Auto0.zip)
I'm not trying to print it, as these are just my settings in a single gcode file.

I'm using the latest bugfix (updated the original description too)

qwewer0, 2020-05-18:
At first it seems to be fixed at the latest 05.18. bugfix.

qwewer0, 2020-06-05:
It seems that the issue is still present, but now I made it to be more general issue.

boelle, 2020-06-21:
@qwewer0 still an issue?

qwewer0, 2020-06-21:
@boelle Yes, M600 is still triggering without it being present in the gcode.

sjasonsmith, 2020-07-04:
I want to check my understanding:

To reproduce this, all you do is "print" your Auto0.g file from the SD card? Repeat until the error occurs?

qwewer0, 2020-07-04:
> To reproduce this, all you do is "print" your Auto0.g file from the SD card? Repeat until the error occurs?

Yes, nothing else.

qwewer0, 2020-07-06:
Same results on Bugfix-2.0.x af8be58d.

qwewer0, 2020-08-01:
Closing this issue, as it seems to be solved on the latest Bugfix-2.0.x 6b0d1549, but this happened already, so we will see.

closed 2020-08-01 by qwewer0