MarlinFirmware/Marlin #18008: [BUG] Cannot select fan 2 in UI

mobilinkd, 2020-05-16:
#7047 # Bug Description

Attempting to select Fan 2 in UI to adjust fan speed selects Fan 3 for adjustment.  I cannot adjust the speed of fan 2 in the UI.

I am using an SKR Pro 1.1 with three fans (2 extruders, 1 part cooling).

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4638563/config.zip)

### Steps to Reproduce

1. Have 3 fans connected to printer.
2. Select Temperature|Fan Speed 2:
3. Note that Fan Speed 3 is selected.

**Expected behavior:** [What you expect to happen]

I expect to be able to control the speed of fan 2.

**Actual behavior:** [What actually happens]

Selecting "Fan Speed 2" takes me to a screen to change to change the speed of fan 3.  And indeed, the speed of fan 3 is changed.  Selecting "Fan Speed 3" in the UI also changes fan 3.

#### Additional Information

I can change fan speed 2 only via gcode, sending M106 P1 S128 for example.

viper93458, 2020-05-17:
I ran into the same issue with my SKR 1.1 pro and reprap discount full graphic smart controller. 

-William

boelle, 2020-06-20:
@mobilinkd still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle