MarlinFirmware/Marlin #18010: [BUG] Upload to Mightyboard fails with new PIO env:MightyBoard2560

grauerfuchs, 2020-05-17:
All right, this might come down to the code maintainers' preferences, but the new env will not upload to any of four tested Mightyboards with the original FlashForge or CTC bootloaders on them. The differences:

Working (old) env:
```
[env:mega2560]
platform          = atmelavr
board             = megaatmega2560
...
upload_speed      = 57600
```
PIO's board config uses the wiring protocol at 115200 for uploads with that "board", so the speed directive had to be added to communicate. However, the new env:
```
[env:MightyBoard2560]
platform          = atmelavr
board             = ATmega2560
...
```
PIO's board config uses the arduino protocol at 115200 for uploads. With only the speed directive added, this still fails to communicate with the bootloader. 

Does it make sense to update the env sections in platformio.ini, adding in the `upload_protocol = wiring` and `upload_speed = 57600` directives, do other manufacturers' boards not have the issue, or is everyone expected to have re-flashed their boards with an Arduino standard bootloader?

ellensp, 2020-05-17:
yip incorrectly added here https://github.com/MarlinFirmware/Marlin/commit/2acdd2fadc0fda6e53ea70957273c7c0f1c931a3

grauerfuchs, 2020-05-17:
I'm certainly not complaining about the fact that it was added; Quite the opposite. I'm glad the new variant is there, because it will make it much easier for people to compile for a board with a design that uses pins that the current Arduino libs can't address without modification. The only issue I have with it is that the upload settings are incorrect for the original bootloader. That's where the question comes in, since solving it for the original is going to make the default settings incompatible with a board that has had the bootloader replaced.

sjasonsmith, 2020-05-17:
My opinion is that if commonly available tools work for a default bootloader then that is how it should be configured. If other options are also potentially used they could possibly be added but commented out.

grauerfuchs, 2020-05-17:
@sjasonsmith : That was my thought as well. However, there are some variations. Trying to get to the bottom of this, I went back and reviewed all the bootloader docs I could find, including the Sailfish configs. Here's what I found from the Sailfish firmware upload xml:

* All the supported boards using the atmega1280 use the STK500v1 protocol at 57600.
     This one should be an easy fix. Under the new env, all we need to do is add the speed directive.

* All but one of the supported boards using the atmega2560 use STK500v2 protocol at 57600. 
* One board (listed as MakerBot Thing-o-Matic (Gen 4) w/ 2560) uses STK500v2 protocol at 115200.
* One listed board uses the atmega644 chip and the STK500v1 protocol at 38400.
     I'm not sure Marlin would run on this board at all regardless of config. I have yet to see a single viable configuration that would fit in the flash space this board has.

grauerfuchs, 2020-05-24:
Update: I've just acquired an FFCP running the bootloader and firmware fresh from the factory. The bootloader header indicates "ATmega2560{null}Arduino Explorer stk500v2 by MLS" and it begins at address 0x3E000. Since the bootloader does indeed use the full space expected, the information in PIO is definitely in error. I'm not sure if this bootloader will allow overwriting of itself, but there is certainly a risk of someone trying to upload too large a file.

closed 2020-05-25 by grauerfuchs