MarlinFirmware/Marlin #18012: [BUG] Must enable THERMAL_PROTECTION_HOTENDS to compile thermal protection code

mobilinkd, 2020-05-17:
### Bug Description

Attempting to disable just THERMAL_PROTECTION_HOTENDS resulted in a compilation failure.

   Marlin/src/module/temperature.cpp:399:69: error: 'WATCH_TEMP_PERIOD' was not declared in this scope; did you mean 'WATCH_BED_TEMP_PERIOD'?

### My Configurations

[config-2.zip](https://github.com/MarlinFirmware/Marlin/files/4639506/config-2.zip)

### Steps to Reproduce

Set the following config options:

    //#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
    #define THERMAL_PROTECTION_BED     // Enable thermal protection for the heated bed
    #define THERMAL_PROTECTION_CHAMBER // Enable thermal protection for the heated chamber

1. Set the above config options
2. Compile code
3. Observe failure

**Expected behavior:** [What you expect to happen]

I expect the code to compile with thermal protection hotends disabled, but bed and chamber protection active.  The MAX31855 code halts when there are spurious "short to GND" errors -- there is no hysteresis for this condition.

**Actual behavior:** [What actually happens]

The code fails to compile.

#### Additional Information

The only workaround is to disable all thermal protection.

sjasonsmith, 2020-05-17:
Thanks for the well-written report. I'd suggest next time including your pins file, since a lot of your enabled features require changes that you must have in there.

I modified your config to compile against bugfix-2.0.x, and disabled a bunch of your options that were missing pins.

Here are some udpated configs which reproduce the error on the latest bugfix.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4639812/Marlin.zip)

I was able to get it to build, but it is late and I am not sure my solution is complete. There seem to be a lot of issues as you try various combinations of PID and protection settings, beyond just the one you found.

mobilinkd, 2020-05-17:
I agree regarding the config issues.  I have hit a few more in testing.

sjasonsmith, 2020-05-24:
@mobilinkd, this should be fixed now. Can you try it out and close this issue if it is resolved?

closed 2020-06-21 by boelle