MarlinFirmware/Marlin #18014: [FR] G-code via I2C bus

DrumClock, 2020-05-17:
Is it possible to send the G-code M25 from the NANO board to the RUMBA board using the I2C bus to pause printing?

[image]

ellensp, 2020-05-17:
I2c doesn't work like that, its a master / slave relation ship. the master send byes to configure and control the slave and the slave responds as needed.  You would have to add routines in Marlin to act as a slave i2c device and design a transfer protocol, then integrate received bytes into marlins command buffers.   None of this exists in marlin at present.  It would be far easier to add a pause on a I/O line. Controlled by a switch or a nano.

DrumClock, 2020-05-17:
Thank you very much for the explanation.
There is a free Endstop Y + on the RUMBA board.
Where do I set up a print pause script in Marlin FW?

ellensp, 2020-05-18:
You have to add code...  Its not a standard option.
Take a look at how the home button is done.  ie search for "HAS_HOME" in Marlin/src/MarlinCore.cpp
Copy that idea... 
Ie setup a pin as input. And later check if its triggered and take action.

DrumClock, 2020-05-18:
Is it possible to use RUMBA analog pins on **EXP3** and **JTAG** connectors as digital I / O pins?
If so, what will be the **pin number for Marlin FW**

[image]

ellensp, 2020-05-18:
All analog pins can be used as digital IO pins.
A9 is Digital pin 36, A10 is Digital pin 35
The Jtag pins are not usable, jtag is enabled by setting fused (hardware switches) in the MPU 
You can technically do this, but its not for beginners. (requires a AVRISP and using avrdude)

DrumClock, 2020-05-18:
Doesn't it have to be 63 and 64?

Grogyan, 2020-05-18:
Keep in mind that I2C/TWI is designed as a intraboard communication bus, and isn't suitable for long connections between boards.

RS485 and CAN Bus are much better, and more resilient against electrical noise generated by the printer and/or nearby sources of electrical noise

closed 2020-05-19 by DrumClock