MarlinFirmware/Marlin #18015: [BUG] RESTORE_LEVELING_AFTER_G28 does not work as expected

rselec, 2020-05-17:
### Bug Description

The Mesh bedleveling isn't ON after starting a print without M420 S when RESTORE_LEVELING_AFTER_G28 is enabled. You can abort the print, goto Motion -> Bed Leveling and its off. Also the printing results proofs that mesh bad leveling wasn't activated.

### My Configurations

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/4640059/Marlin_config.zip)

### Steps to Reproduce

1. Use and Compile Marlin 2.0.5.3 with attached configuartion (almost stock, expect mesh bed leveling, and scrolling long file names) for Anet A8
2. Perform mesh bed leveling and store settings to EEPROM
3. Slice an printing object without the M420 S G-Code (the RESTORE_LEVELING_AFTER_G28 should do the job)
4. Powercycle the printer and print from sdcard

**Expected behavior:** [What you expect to happen]

Mesh bed leveling should be activated after homing G28.

**Actual behavior:** [What actually happens]

Mesh bed leveling is off during print and also after aborting or finishing the print.

#### Additional Information

I added the M420 S to the end of my machine code start procedure, with the code mesh bed leveling is turned on and working. But this command shouldn't be necessary to my understanding.

GMagician, 2020-05-17:
restore leveling should restore mesh state home, not activate it after home

bill-orange, 2020-05-17:
@GMagician are you sure about that?  Marlins description of M420 states:

G28 disables bed leveling. Follow with M420 S to turn leveling on, or use RESTORE_LEVELING_AFTER_G28 **to automatically keep leveling on** after G28.

bold font is my edit.

sjasonsmith, 2020-05-17:
I believe @GMagician is correct. It simply avoids the confusion when G28 turns off leveling, but I don’t think it is expected to enable it if previously disabled.

GMagician, 2020-05-17:
> are you sure about that? Marlins description of M420 states:

Yes I'm sure, here is code:
`TERN_(RESTORE_LEVELING_AFTER_G28, const bool leveling_was_active = planner.leveling_active);`
this saves current state and below restore after home
`  TERN_(RESTORE_LEVELING_AFTER_G28, set_bed_leveling_enabled(leveling_was_active));`

bill-orange, 2020-05-17:
@GMagician I am convinced.

rselec, 2020-05-17:
Ok, so this isn't a bug. But the description of the command here is misleading: https://marlinfw.org/docs/gcode/M420.html
And even more the explanation is this video is wrong or at least incomplete: https://www.youtube.com/watch?v=38PkynA1uGI

So that means for me, that I have to insert the M420 S command somewhere in the start G-code of cura.
If I enabled RESTORE_LEVELING_AFTER_G28 it doesn't matter if it is before or after G28 commands.
If not, the M420 S must be after the last G28 command to have an effect.

Is there any possibility to enable the mesh bedleveling when I don't have M420 S in the G-code of the file? My Anet A8 does not offer the enable option via LCD before I abortet the first try. Than it is displayed as off and I could switch it on...
Maybe that is another I don't get right or it is a bug?

Thanks for your support!

closed 2020-05-17 by rselec

GMagician, 2020-05-17:
@rselec I read document, yes is not completely clear but a very little thing may rise suspect:

> to turn leveling on, or use RESTORE_LEVELING_AFTER_G28 to automatically keep leveling on after G28.

it states **keep** leveling on, not activate leveling, but I think message needs something that leaves no doubt

rselec, 2020-05-17:
Yes, now since I know it...

Any idea about the LCD menu item?

GMagician, 2020-05-17:
> Any idea about the LCD menu item?

I think I missed this. There is a item to toggle but is not always present.
`AUTO_BED_LEVELING_UBL` and `LCD_BED_LEVELING` must be disabled and
`HAS_LEVELING` and `SLIM_LCD_MENUS` disabled. So depending on your configuration it may be present or not. If present you need axis homed and a valid leveling data....

GMagician, 2020-05-17:
I checkd... then:
when AUTO_BED_LEVELING_UBL is enabled:
you have an item that executes G29 A or G29D messages are 'Activate UBL' or 'Deactivate UBL'
otherwise:
you have an item that toggle and message is 'Bed Leveling' with state but only when homed and valid mesh

rselec, 2020-05-17:
**Only when homed** is the point.

So I have to manually perform an auto home after power on, then the option to enable bed leveling appears and I can turn it on.
After that I could start a print without M420 S command but with active mesh bed leveling.

This is not was I expected in the morning but now I know how it works and I'm fine with that!

Many thanks for your explanations!

Roxy-3D, 2020-05-18:
> @GMagician I am convinced.

There is a lot of history here.   The original 3-Point and then Grid bed leveling disabled bed leveling after a G28.   The main reason for doing this was G28 needed bed leveling disabled in order to do its work.    

There was a difference of opinion whether that made sense or not.   When UBL was written, G28 preserved the bed leveling state upon entry and restored it at the end.    Understandably, that difference wasn't accepted by everybody.    The compromise was to make it optional (for all bed leveling systems) whether to preserve the bed leveling state at entry of the G28 command.

Now, with the current options, everybody can pick and choose what they are comfortable with and set the machine up to behave however they want it to behave.