MarlinFirmware/Marlin #18016: Error al compilar

rp8020, 2020-05-17:
Hola, tengo una reprap de Anet 8, quiero instalarle el Marlin 2.0 y he visto varios videos, pero al poner mis parametros en el Marlin 2.0.5 buggins e intentar compilar me sale el siguiente mensaje de error:
Arduino:1.8.5 (Windows 10), Tarjeta:"Anet V1.0"

Archiving built core (caching) in: C:\Users\rp802\AppData\Local\Temp\arduino_cache_933581\core\core_anet_avr_anet_eebe80b08784ba478cff07c27703f803.a
C:\Users\rp802\AppData\Local\Temp\ccROE7uh.ltrans0.ltrans.o: In function `main':

C:\arduino-1.8.5\hardware\arduino\avr\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\arduino-1.8.5\hardware\arduino\avr\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compilando para la tarjeta Anet V1.0.

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

No se como solucionarlo, me podeis dar una ayuda.

Gracias

boelle, 2020-06-20:
Writing in english works better

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle