MarlinFirmware/Marlin #18019: bugfix 2.0.x 17-05-20 won't compile with i2cbus enabled.

jifop, 2020-05-17:
I cannot get the latest bugfix to compile unless I disable experimental i2c Bus however I need this enabled to control my fan relay via gcode as I cannot work out how to integrate it into marlin.

Compiles fine with experimental i2cbus disabled.

Errors below:
Marlin\src\feature\twibus.cpp: In member function 'bool TWIBus::request(uint8_t)':

Marlin\src\feature\twibus.cpp:107:48: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:

107 | if (Wire.requestFrom(I2C_ADDRESS(addr), bytes) == 0) {

| ^

In file included from Marlin\src\feature\twibus.h:26,

from Marlin\src\feature\twibus.cpp:27:

C:\Users\Tim\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:51:13: note: candidate 1: 'uint8_t TwoWire::requestFrom(int, int)'

51 | uint8_t requestFrom(int, int);

| ^~~~~~~~~~~

C:\Users\Tim\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Wire.h:50:13: note: candidate 2: 'uint8_t TwoWire::requestFrom(uint8_t, uint8_t)'

50 | uint8_t requestFrom(uint8_t, uint8_t);

| ^~~~~~~~~~~

Marlin\src\feature\twibus.cpp:108:44: error: call of overloaded 'debug(const char [13], int)' is ambiguous

108 | debug("request fail", I2C_ADDRESS(addr));

| ^

In file included from Marlin\src\feature\twibus.cpp:27:

Marlin\src\feature\twibus.h:236:26: note: candidate: 'static void TWIBus::debug(const char*, uint32_t)'

236 | static inline void debug(const char[], uint32_t) {}

| ^~~~~

Marlin\src\feature\twibus.h:237:26: note: candidate: 'static void TWIBus::debug(const char*, char)'

237 | static inline void debug(const char[], char) {}

| ^~~~~

Marlin\src\feature\twibus.h:239:26: note: candidate: 'static void TWIBus::debug(const char*, uint8_t)'

239 | static inline void debug(const char[], uint8_t) {}

| ^~~~~

*** [.pio\build\LPC1769\src\src\feature\twibus.cpp.o] Error 1

sjasonsmith, 2020-05-17:
Please attach your configuration files as requested in the issues template. This makes it easier for somebody to try to help you.

jifop, 2020-05-17:
> Please attach your configuration files as requested in the issues template. This makes it easier for somebody to try to help you.

apologies, have attached them

sjasonsmith, 2020-05-19:
Thanks! I've posted a pull request which should resolve the issue. The change is trivial if you want to implement it your your own copy prior to it being merged in.

jifop, 2020-05-19:
That would be amazing thankyou

closed 2020-06-20 by boelle