MarlinFirmware/Marlin #18020: [BUG] 4th Place Digit or higher cut off on Velocity Screen

rflulling, 2020-05-17:
### Bug Description
On the monochromatic LCD, on the velocity menu, only 3 digits are shown, the 4th is cut off. 
[Marlin Config BF2053.zip](https://github.com/MarlinFirmware/Marlin/files/4640578/Marlin.Config.BF2053.zip)

### My Configurations

[Marlin Config BF2053.zip](https://github.com/MarlinFirmware/Marlin/files/4640578/Marlin.Config.BF2053.zip)

### Steps to Reproduce

Not sure how much info you need here. 
**Configuration -> Advanced settings -> Velocity**
Observe that any value greater than 999 will be cut off. 

#### Additional Information

Potentially related?
I have also noticed that numbers on other screen are either Rounded up, or will refuse to configure. For example Acceleration Max E1/E2 may be set in configuration.h or using M201 T0 E80. However using the job dial it is not possible to set a value less than 100. Similarly in Jerk settings, it is possible to set a value in firmware or send as command, but it is not possible to set these same value with the jog dial. -I do not believe there are any notes explaining this as expected.

boelle, 2020-06-20:
@rflulling still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

rflulling, 2020-06-25:
@boelle 
The bugfix that I am running this issue seems to be resolved.

I will look forward to the merge to determine if this requires another review.