MarlinFirmware/Marlin #18021: [BUG] #define PID_AUTOTUNE_MENU results in incorrect menu item

bill-orange, 2020-05-17:
Bug Description:
Bug Description:
Enabling PID_AUTOTUNE_MENU results in the requested menu, in the proper spot.  However, PID Autotune E1 is offered when only one extruder ,E0 exists.

My Configurations: included at end of post

### Steps to Reproduce

1. Scroll through to the autotune menu correctly located in Configuration -> Advanced Configuration -> Temperature
2. Observe last line. Autotune for E1 is offered onstead of E0

Expected behavior: Menu item should be PID Autotune E0

Actual behavior: Menu item should be PID Autotune E1

#### Additional Information
 I am assuming that the number displayed after PID Autotune E1 is the starting temperature or target temperature.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4640846/Marlin.zip)
[image]

boelle, 2020-06-20:
@bill-orange still an issue?

bill-orange, 2020-06-21:
I think we can close it

closed 2020-06-21 by bill-orange