MarlinFirmware/Marlin #18022: skr pro V1.1 endstops not working

carl667, 2020-05-17:
hello I recently installed the skr pro v1.1 into my ender5 plus i can get everything working on in execpt for the endstop on the x and y the z probe works but  it doesn't stop you can still raise the bed after its been triggered I tried inverting endstops still same result im using tmc2130 without spi. When I trigger the endstops the lights on the board activate. currently using the ender5 profile from the config folder as there is not one for the 5plus.

swilkens, 2020-05-17:
Please fill out the new issue form you are presented with when you open a new issue.

Provide at least:

1. Configuration files (configuration.h and configuration_adv.h)
2. Printer make / model
3. The version of the firmware you are using
4. Try latest bugfix-2.0.x to see if you also have the issue with that version.

carl667, 2020-05-17:
[5+config.zip](https://github.com/MarlinFirmware/Marlin/files/4640987/5%2Bconfig.zip)
marlin 2.0.5.3 current version using 
example config used released config ender 5 pro 
machine Ender 5 plus with th3d ABL 
tried latest bug fix still same issue running proterface m113 x,y stay triggered, z will say open and triggered when triggered but the bed will still move after it has been triggered I have also tried inverting the endstops still same result.

fiferox, 2020-05-18:
I erase the eeprom and rewrite marlin on my skr mini e3 1.2, and now this bug disapear.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle