MarlinFirmware/Marlin #18024: [BUG] LCD screen flickering characters bug

Horizongaming1, 2020-05-17:
### Bug Description

My LCD screen flickers in weird chinese/ASCII characters
https://www.youtube.com/watch?v=eOpBX6If8AI
(the display is switching through several "sites" full of weird characters, even if I'll not turn the knob)

### My Configurations
My Printer:
  Ender-3 (Mainboard swapped to MKS Gen. 1.4 ATmega 2560 RAMPS1.4)

LCD Screen:
  Standard Creality CR-10 Display

Here are my configuration files:
  https://drive.google.com/file/d/19WJT6BvKEe-3cF8J9cOfQ8prhsxmRp6j/view

Marlin Version 2.0

### Steps to Reproduce

So I edited the configuration files after the templates from the Marlin Github page for my Ender-3.
But this is not the first time I do this, because there was an issue with the Raspberry Pi program "Klipper". This program flashes the ATMega for "performance improvements". But there was a Problem with the program so I decided to reflash the original Marlin firmware back on the chip but the mainboard/LCD wouldn't work anymore.
I hope I didn't brick my Mainboard.

**Expected behavior:** So I expected that the Mainboard/LCD works fine.

**Actual behavior:** But the LCD shows weird graphics.

### Sorry for my bad english

boelle, 2020-06-20:
@Horizongaming1 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle