MarlinFirmware/Marlin #18026: [BUG] STARTUP_COMMANDS Not Parcing Multiple G-Codes

borland1, 2020-05-18:
In Configuration_adv.h file, Startup Commands section provides sample format for only a single g-code:
//#define STARTUP_COMMANDS "M17 Z"

When sending multiple g-codes, parcer seems to ignore subsequent g-codes.  Sending g-code M300 to play series of tones on powerup with the following configuration, plays only one short tone.

// Play Startup Tones ...
#define STARTUP_COMMANDS "M300 S440 P200, M300 S660 P250, M300 S880 P300"

### My Configurations
[2.0.5.3_ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/4641500/2.0.5.3_ConfigFiles.zip)

ellensp, 2020-05-18:
The specification for gcode is is one gcode per line. You need to insert \n between each gcodes so that the string is interpreted as multiple lines and not just one line.

borland1, 2020-05-18:
ellensp, Thank you.

This tests good...

#define STARTUP_COMMANDS "M300 S440 P200 \n M300 S660 P250 \n M300 S880 P300"

closed 2020-05-18 by borland1