MarlinFirmware/Marlin #18027: [FR] Option for cooling fan enable on linear advance Calibration pattern.

picclock, 2020-05-18:
If possible :-)
Include an option for cooling fan enable on linear advance Calibration pattern generator.
https://marlinfw.org/tools/lin_advance/k-factor.html
Just a tickbox to enable the cooling fan. M106 start M107 stop.
Many Thanks
Best Regards
picclock

### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

qwewer0, 2020-05-18:
Why would you want cooling for that?

picclock, 2020-05-18:
Because if the fan is turned off the filament strings everywhere. Ender 3
btw, pla silk filament.

Many thx

Picclock

qwewer0, 2020-09-27:
@boelle 
This FR can be closed, as the LA Calibration page now has Fan Speed (%) option.

closed 2020-09-27 by thisiskeithb