MarlinFirmware/Marlin #18028: [BUG] G28 with BLTouch does not move to XY coordinates - CoreXY

albert2004, 2020-05-18:
### Bug Description
CoreXY, when issuing G28, printer moves to X=0, Y=0 and probes Z in this position, but it should with offset. In my case: X=42, Y=13.
In addition, X and Y is set to X=42 and Y=13, but physically it is 0,0

Issue is valid when X is less than 42.

### My Configurations
Marlin bugfix-2.0.x branch, hash 213d4b890
[Config_213d4b890.zip](https://github.com/MarlinFirmware/Marlin/files/4643456/Config_213d4b890.zip)

### Steps to Reproduce
Works incorrectly when issue like that:
G28
or
G28 XY
G28 Z

or
G28 XY
G1 X41 - value less than expected X for probing
G28 Z

Works correctly when issue like that:
G28 XY
G1 X42 - to value greater or equal to probe offset + probe margin
G28 Z  - now, carriage will move to correct Y position

**Expected behavior:**
Move probe to correct X,Y position.
Update X,Y position according to physical position.

**Actual behavior:**
Z probe is made at physical position of 0,0
Reported X,Y position is incorrect - shows desired probing position

#### Additional Information
Logs from G28 debugging, in marked area, there is no physical movement of the printer:
Send: G28
[...]
Recv: >>> G28
Recv: Machine Type: CoreCartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X-32.00 Y17.00 Z-1.53 (Left-Back & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: Raise Z (before homing) to 4.00
Recv: >>> do_blocking_move_to X0.00 Y0.00 Z4.00
Recv: <<< do_blocking_move_to
[...]
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -432.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00)
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: Axis 88 home_offset = 0.00 position_shift = 0.00
Recv: > home_offset[X] = 0.00
Recv:   current_position= X0.00 Y0.00 Z4.00 :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position= X0.00 Y0.00 Z4.00 : sync_plan_position
Recv:   current_position= X0.00 Y0.00 Z4.00 : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -396.00, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 25.00)
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: Axis 89 home_offset = 0.00 position_shift = 0.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position= X0.00 Y0.00 Z4.00 :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position= X0.00 Y0.00 Z4.00 : sync_plan_position
Recv:   current_position= X0.00 Y0.00 Z4.00 : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: BLTouch Command :160
Recv: BLTouch Command :90
Recv: last_written_mode - 0
Recv: config mode - OD
Recv: home_z_safely >>>
Recv:   current_position= X0.00 Y0.00 Z4.00 : sync_plan_position
**Recv:   destination= X42.00 Y13.00 Z5.53 : home_z_safely
Recv: >>> do_blocking_move_to X42.00 Y13.00 Z4.00
Recv: <<< do_blocking_move_to**
Recv: echo:busy: processing
[...]
Recv: >>> homeaxis(Z)
Recv:   current_position= X42.00 Y13.00 Z4.00 : Probe::set_deployed
Recv: deploy: 1
Recv: Probe::move_z(4.00)
Recv: >>> do_blocking_move_to X42.00 Y13.00 Z5.53
Recv: <<< do_blocking_move_to
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X42.00 Y13.00 Z5.53
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_homing_move(Z, -525.00, [4.00])
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(Z)
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv: Move Away:
Recv: >>> do_homing_move(Z, 4.00, 4.00)
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_homing_move(Z, -8.00, 2.00)
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(Z)
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv: >>> set_axis_is_at_home(Z)
Recv: *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
Recv: > probe.offset.z = -1.53
Recv: Axis 90 home_offset = 0.00 position_shift = 0.00
Recv: > home_offset[Z] = 0.00
Recv:   current_position= X42.00 Y13.00 Z1.53 :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position= X42.00 Y13.00 Z1.53 : sync_plan_position
Recv:   current_position= X42.00 Y13.00 Z1.53 : > AFTER set_axis_is_at_home
Recv:   current_position= X42.00 Y13.00 Z1.53 : Probe::set_deployed
Recv: deploy: 0
Recv: Probe::move_z(4.00)
Recv: >>> do_blocking_move_to X42.00 Y13.00 Z5.53
Recv: <<< do_blocking_move_to
[...]
Recv: echo:busy: processing
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv: >>> do_blocking_move_to X42.00 Y13.00 Z5.53
Recv: <<< do_blocking_move_to
Recv: <<< homeaxis(Z)
Recv: <<< home_z_safely
Recv:   current_position= X42.00 Y13.00 Z5.53 : sync_plan_position
Recv: X:42.00 Y:13.00 Z:5.53 E:0.00 Count A:4400 B:2320 Z:8848
Recv: <<< G28
Recv: ok
Send: M113 S2
Recv: ok

albert2004, 2020-05-18:
This same goes to G29. When probing and X,Y coordinates goes for X below minimum X for probe then next moves will not be done.

albert2004, 2020-05-21:
Root cause found.
In the configuration I have enabled USE_XMAX_PLUG without connecting correct endstop.
It results in not accepting to go X++ (greater than current position), for G28 and G29, since endstop was constantly active.
Please review this behavior. If this is HW issue then I will close issue, but I believe:
- there was no debug information about hitting enstop - this would help with investigation
- normal moves are not affected - I found that G28 and G29 are affected by XMAX endstop.

boelle, 2020-06-20:
@albert2004 still an issue?

albert2004, 2020-06-21:
@boelle no, I left this open for maintainers to decide if not informing about incorrect move (caused by not connected endstop) should be somehow handled (for example debug print that device cannot move to desired position) or it should be left as it is.

closed 2020-06-23 by boelle