MarlinFirmware/Marlin #18033: [BUG] ABL - Not Probing all the points

psavva, 2020-05-18:
### Bug Description

I have enabled Auto Bed Level Bilinear and found that not all points that are supposed to be probed are being probed.

I have a 7 x 7 mesh whish should probe 49 points (which is  accurately displayed on the LCD), however, only the first row is being probed, ie: the 1st 7 points are being probed, and no more.

### My Configurations

Configuration : https://github.com/psavva/Marlin/blob/9f935383099ec295145b53cfb9a3775854144dff/Marlin/Configuration.h
Configuration adv: 
https://github.com/psavva/Marlin/blob/9f935383099ec295145b53cfb9a3775854144dff/Marlin/Configuration_adv.h

### Steps to Reproduce

1. G28
2. G29

**Expected behavior:** Probing of all Points should occur

**Actual behavior:** Probing stops after the 1st 7 points.

#### Additional Information

Uploading Video on Youtube which you can see the machine in action.

psavva, 2020-05-18:
I will also provide the following tomorrow:

- Download Marlin bugfix-2.0.x to test with the latest code.
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Issue M111 S247 to enable maximum logging.
- Issue G28
- Issue G29
- Attach the log output into a .TXT file

psavva, 2020-05-19:
I see that lots of changes made it into fac3a7d14788d7f80112d0c6456ae460b18764b7 yesterday.
I will retest shortly based on the latest as of now, and report back with results

psavva, 2020-05-19:
Debug Info after latest update and enabling debug_leveling_feature, and m1114_detail, with maximum logging enabled

[G29-Probes 7 out of 49 only - debug info.TXT](https://github.com/MarlinFirmware/Marlin/files/4648357/G29-Probes.7.out.of.49.only.-.debug.info.TXT)

psavva, 2020-05-19:
I have noticed this in the log:  
`08:12:33.887 : Error:Probing Failed`

I think it's only returned by probe.cpp lines 709 - 731
`  float measured_z = NAN;
  if (!deploy()) measured_z = run_z_probe(sanity_check) + offset.z;
  if (!isnan(measured_z)) {
    const bool big_raise = raise_after == PROBE_PT_BIG_RAISE;
    if (big_raise || raise_after == PROBE_PT_RAISE)
      do_blocking_move_to_z(current_position.z + (big_raise ? 25 : Z_CLEARANCE_BETWEEN_PROBES), MMM_TO_MMS(Z_PROBE_SPEED_FAST));
    else if (raise_after == PROBE_PT_STOW)
      if (stow()) measured_z = NAN;   // Error on stow?

    if (verbose_level > 2) {
      SERIAL_ECHOPAIR_F("Bed X: ", LOGICAL_X_POSITION(rx), 3);
      SERIAL_ECHOPAIR_F(   " Y: ", LOGICAL_Y_POSITION(ry), 3);
      SERIAL_ECHOLNPAIR_F( " Z: ", measured_z, 3);
    }
  }

  feedrate_mm_s = old_feedrate_mm_s;

  if (isnan(measured_z)) {
    stow();
    LCD_MESSAGEPGM(MSG_LCD_PROBING_FAILED);
    SERIAL_ERROR_MSG(STR_ERR_PROBING_FAILED);
  }`

psavva, 2020-05-19:
This is the log with G29 V4 
For Verbosity Logging

[G29-Probes 7 out of 49 only - G29 V4 - debug info.TXT](https://github.com/MarlinFirmware/Marlin/files/4649926/G29-Probes.7.out.of.49.only.-.G29.V4.-.debug.info.TXT)

psavva, 2020-05-19:
This is the log with G29 V4 and 
#define BLTOUCH_DELAY 500

[G29-Probes 7 out of 49 only - G29 V4 -BLTOUCH_DELAY-500- debug info.TXT](https://github.com/MarlinFirmware/Marlin/files/4649939/G29-Probes.7.out.of.49.only.-.G29.V4.-BLTOUCH_DELAY-500-.debug.info.TXT)

psavva, 2020-05-19:
As Another Test, I've disabled Multiple Probing:
//#define MULTIPLE_PROBING 2
//#define EXTRA_PROBING    1

Now Probing has completed all 49 Points.

Seems there is an issue with Extra Probing...? 
[G29-Probes 7 out of 49 only - G29 V4 -BLTOUCH_DELAY-500- debug info.TXT](https://github.com/MarlinFirmware/Marlin/files/4649981/G29-Probes.7.out.of.49.only.-.G29.V4.-BLTOUCH_DELAY-500-.debug.info.TXT)

psavva, 2020-05-19:
Update: I've switched to UBL and for around 70 probe points, not one Probing Failure as reported by Bilinear Bed Leveling.  Seems the issue can be narrowed down to other bed levelling systems...

sjasonsmith, 2020-05-29:
The important line from your logs is this one:

> 13:55:08.413 : SLOW Probe fail! - Triggered early.

This is related to a sanity check that is a recent addition to Marlin. It attempts to detect unexpected early probes, and fails probing to avoid damaging your printer.

The behavior of this sanity check is influenced by `Z_CLEARANCE_BETWEEN_PROBES `. You could try increasing that value (perhaps from 5 to 10), to see if it makes your issues go away. If so, there may be an issue with that logic, or your bed may be very non-level, causing one the probe to trigger much earlier than expected.

psavva, 2020-05-29:
Does UBL not take Z_CLEARANCE_BETWEEN_PROBES into consideration?

UBL is working perfectly.

boelle, 2020-06-21:
@psavva still an issue?

psavva, 2020-06-21:
Hi @boelle 

I believe that there is a bug here.
As I was not able to solve it, I've moved over to UBL which I'm just with.  Probing is working perfectly with UBL whereas probing fails using ABL...

Others have faced this same issue and closed it as a duplicate based on the fact that this issue is open.

I recommend that one of the devs which understand the probing code better than I do review why probing fails under ABL but is just fine under UBL.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle