MarlinFirmware/Marlin #18035: [FR] Make DGUS_LCD_UI_ORIGIN config. coherent to Configuration*.h

rubienr, 2020-05-18:
### Description

As a user of Marlin firmware having a DGUS display, I want to use `DGUS_LCD_UI_ORIGIN` without the need to touch the `/dgus/origin/DGUSDisplayDef.h/cpp`. The solely point of configuration shall be `Configuration.h` and `Configuration_adv.h`).

### Feature Workflow

1. set up `Configuration.h` and `Configuration_adv.h` 
2. compile MarlinFirmware
3. flash & go

### Feature Workflow Now

1. set up `Configuration.h` and `Configuration_adv.h` 
2. compile MarlinFirmware
3. fix compiler issues in `/dgus/origin/DGUSDisplayDef.h/cpp`
4. configure display, example: un-/comment code to reflect the number of extruder
2. compile MarlinFirmware
3. flash & go

#### Additional Information

I found https://github.com/MarlinFirmware/Marlin/pull/17142 to be related but my concern does not fit 100% there. There is barely overlap with  https://github.com/MarlinFirmware/Marlin/pull/17142 and a merge could be done independently.

I spotted the `dgus/origin/DGUSDisplayDef.h/cpp` could use refactoring because
* I saw lot of dead/unused code,
* the configuration must be done by un-/commenting lines in code instead via `Configuration*.h`, and
* minor issues found.

Basically I am **asking for a brief review** of [this branch](https://github.com/rubienr/MarlinFirmware/commits/2.0.x-extui-dgus-origin), to see if my changes are going the correct way and could make it once to upstream.

##### What was achieved

As a starting point I took `/dgus/origin/DGUSDisplayDef.h/cpp` for refactoring and could refine it so that no un-/commenting lines is necessary any more. Those lines are now depending on `Configuration*.h`. Furthermore some missing parts were added: mostly related to extruder 2-5 variable pointer. I tried to apply the "do not harm principle" so no addresses have been modified to stay backward compatible, except of the boot screen ID.

An additionally driver is the fact that Creality did not open the sources of the Ender-5 Plus that applies a DGUS display. For this reason I started to enhance work done by [coldtobi](https://github.com/coldtobi/Marlin_DGUS_Resources) and created a new [DGUS project](https://github.com/rubienr/MarlinDgusResources/) with the aim of being
* highly configurable, and 
* auto generated as far as possible.
 
##### Other issues found

1. IMHO boot screen ID should not reside on `0x0` since that is the starting address of the default font file. I think that does not cause issues since it only overwrites the first 400kB of the 3MB font file. For this reason I changed it from `0` to `12` which introduces an incompatibility.
2. I think there is a misunderstanding w.r.t. VP (value pointer). As far I understand DGUS addresses words (2-bytes) rather than 1-byte. Now many `uint16_t` VP in `/dgus/origin/DGUSDisplayDef.h` have a spacing of `2` which I find is wrong.
3. On press a new active screen was not requested by Marlin from the display to be shown. This was latent due to questionable display settings 
https://github.com/rubienr/MarlinFirmware/commit/e36730827ca982c8a09f01d58f2d1feea857f9c0

pedrovanzella, 2020-07-19:
From what I can see in your branch, not everything is properly gated by `ENABLED(PSU_CONTROL)` and `ENABLED(CASE_LIGHT_ENABLED)`.

Chasing the issue down and gating `const uint16_t VPList_Psu[] PROGMEM` and `const uint16_t VPList_Lights[] PROGMEM` leads to more and more compilation issues. Eventually I was able to get the firmware to compile by ungating `uint16_t swap16(const uint16_t value)`, but at that point I wasn't sure anymore I was even compiling something that made any real sense.

It's not really clear to me what your branch adds, actually. Your other project says it relies on these changes to work properly, but what exactly doesn't work without them?

From the commit messages, it looks like your branch does a few different things, such as adding functionality (PSU Control, UBL), to refactoring the code. Sending separate PRs will probably make the changes easier to test and digest and increase the chances we see them merged.

rubienr, 2020-07-21:
@pedrovanzella You are right. I lost track of this FR at some point, although I had in mind that it shall be closed because it is obsolete. I also digged into the code, but then I decided to refactor the DGUS implementation. I suggest to close this FR and apologize for the extra effort I introduced. Later I will send a new PR or even better a review request with my changes. I achieved a lot but now I am at the last 20% which makes 80% of the effort (and my printer is temporarily not functional for testing).

Sebazzz, 2020-11-25:
> IMHO boot screen should not reside on ID=0x0 since that is the starting block of the default font file on DGUS' flash storage. I think that issues is latent since it only overwrites the first 400kB of the 3MB font file (or I did not understand the sparse DGUS documentation correctly). For this reason the ID was changed from 0 to 12 which introduces an incompatibility. Btw. if the hiprecy code works, it contradicts my theory (uses IDs: 1, 3, 7, 9).

Are you positive about that? The DWIN screen IDs do not correlate to the memory locations of the files.

Why not? Because of the following:

- The numbers of the files loaded into DWIN determine the memory location (0_DWIN_ASC.HZK, 13TouchFile, 32_Screen).
- The screens are not separate bitmaps - they are compiled into an ICL file. Within that ICL file you have the index so you can address the specific "icon" (= background/screen) in the ICL file.

rubienr, 2020-11-27:
Hello,
No I am not 100% sure. Can you provide a link to documentation about this? Tried to read a lot I found in the net but the documentation is not very accurate. Until now I was assuming the number prefix determines the memory block the file is placed in.

Sebazzz, 2020-11-27:
> 
> 
> Hello,
> No I am not 100% sure. Can you provide a link to documentation about this? Tried to read a lot I found in the net but the documentation is not very accurate. Until now I was assuming the number prefix determines the memory block the file is placed in.

Yes it does, but DWIN doesn't use JPEG files nor BMP files. You compile the screens into an ICL file, which has indeed that prefix (which is not 0). In that ICL file, which contains multiple images, you address the images by index *in* that ICL file. 

I have sourced some documentation here: https://github.com/CR6Community/CR-6-touchscreen#documentation-for-development

thinkyhead, 2021-12-29:
Display code for the E3V2 / DWIN controller are continually being refined, and gradually configuration options are being applied to these displays. With that in mind, we don't need a specific issue about it, so this is being closed.

closed 2021-12-29 by thinkyhead