MarlinFirmware/Marlin #18037: [BUG] (short description)

visualcortexlab, 2020-05-18:
### Bug Description

ok I'm giving up. I can't get this to work. :((
after days of googling, youtubing and little sleep I'm giving up and hopefully someone here can see the problem.
Sensorless homing not working.
As soon as I issue a G28 the printer sets (0,0,0) wherever the extruder is at that time without making a single move.
Homing Z works, which means that most of the time I also crash both X and Y because they're not Homed.
in Marlin I tried changing all sort, voltage, sensitivity, anything. Not a single change in behaviour. So I must be missing something very obvious, and its been days of the same exact failure, over and over.
Some help is really really appreciated as I have my printer all over the table since 2 two weeks trying to configure it :\
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4646936/configs.zip)

Thank you for any help!

ellensp, 2020-05-19:
This is not a general support forum.
Please read https://github.com/MarlinFirmware/Marlin/issues/14694 and use one or multiple of those services.

swilkens, 2020-05-21:
sensorless homing can be difficult to tune, similar issues and tuning approach described here:

https://github.com/MarlinFirmware/Marlin/issues/15848

closed 2020-06-21 by boelle