MarlinFirmware/Marlin #18038: SD card change corrupts printer settings

ModMike, 2020-05-18:
### Bug Description

Changing the SD card cases printer to lose it's position. Subsequent homing causes x-axis to shift over about 300mm and crash into Max X. Z motor also makes loud humming sound and jerks suddenly.

### My Configurations

Ender 3 Pro
Fysetc Cheetah 1.1b TMC2209 (new version)
Marlin 2.04.4 (tried all version right up to bug fix, same issue) 

### Steps to Reproduce

1) Turn printer on
2) Remove SD
3) Write new print file
4) Insert SD Card
5) Auto Home

Z axis will make a loud sound when starting, as if it was stuck and suddenly let go. Motor will hum loudly. The printer will then try to home but X will start 30mm or more to the right and crash into X max position. If not restarted the printer grinds. 

#### Additional Information

This issue has been happening ever since I upgraded from Marlin 2.0 bug fix that was last updated about 6 months ago.

- I refactored the Fysetc Cheetah config.h and config_adv.h for latest Marlin and current version I am using. Where used without changes and same issue
- I created a fresh config, with no changes to the settings, same issue
- The currently attached configs print perfectly but have many features enabled, SD card change still causes an issue.
- I did not INIT the card at any time as the printer recognized when the card was ejected and loaded
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4647305/configs.zip)

boelle, 2020-06-20:
@ModMike still an issue?

ModMike, 2020-06-21:
It’s fixed in latest bug fix so I will close it.

closed 2020-06-21 by ModMike