MarlinFirmware/Marlin #18041: Cannot reset MAXTEMP error shutdown loop

mjp808, 2020-05-19:
BOARD: Brand new tested SKR 1.4
After checking all hardware good, trying to flash clean Marlin, MAXBED TEMP error and forced restart persistent loop, no way to clear error All wires removed, clean Marlin flashed via SD in 5v USB serial mode Pronterface.

Request some way to drop a factory reset define in with firmware

ellensp, 2020-05-19:
It a micro controller, flash is over written with new firmware...   errors are not kept across firmware updates. Something else is wrong.

mjp808, 2020-05-19:
I made a mistake in OP, the error is e1 maxtemp, but essentially same issue
his is the second board thats stuck in this loop. The only way to get access to the board is giving temp sensor 0 a dummy 998 temp. m502/m500 doesnt clear it. If there was a way of flushing out old temp compensation settings it would help. thermistors are all tested and working.

ellensp, 2020-05-26:
Since you have now discovered the short from the heater cartage to the thermistor pins, instantly annihilating the controller. Please close this issue. as Its not an firmware issue.

closed 2020-05-26 by mjp808

mjp808, 2020-05-26:
Not %100 sure thats the problem, but will do