MarlinFirmware/Marlin #18042: SENSORLESS_PROBING compile failure with TMC_2209

hackebike, 2020-05-19:
I am trying to compile marlin from today and I am getting the following compiling error:
`In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:40:0,
                 from Marlin/src/HAL/STM32/timers.h:25,
                 from Marlin/src/HAL/STM32/HAL.h:121,
                 from Marlin/src/HAL/STM32/HAL.cpp:25:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2387:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."`

I defined SENSORLESS_PROBING and using TMC_2209 in my configs

Any idea?
thx

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4649115/configs.zip)

ellensp, 2020-05-19:
disable both of these
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING

hackebike, 2020-05-19:
Thanks that fix the problem. So I am guessing there were less checks before since it was compiling fine several months ago. Good to know.

closed 2020-05-19 by hackebike