MarlinFirmware/Marlin #18044: [BUG] Screen throttling can't handle fast displays

jmz52, 2020-05-19:
### Bug Description

Display with update time less than 1 ms suffers from screen throttling.
uint16_t max_display_update_time is 0 due to really fast display update time.
`max_display_update_time--`  set max_display_update_time to 65535, meaning next screen update will be in 18 hours.

### My Configurations

MKS Robin with reworked LCD subsystem

### Steps to Reproduce

None.

**Expected behavior:** [What you expect to happen]
UI is displayed when printing from SD card

**Actual behavior:** [What actually happens]
Blank screen when printing from SD card

#### Additional Information

Possible solutions:
- Replace `max_display_update_time--;` with `if (max_display_update_time) max_display_update_time--;` in ultralcd.cpp (tested)
- Change max_display_update_time type to int16_t (not tested)
- Add 1 ms delay to MarlinUI::draw_status_screen() (tested)

GMagician, 2020-05-19:
Where do you see `max_display_update_time--` I can't see it in bugfix 2.0.x

jmz52, 2020-05-19:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/ultralcd.cpp#L970

GMagician, 2020-05-19:
Download bugfix 2.0...it has been changed and there is no more such line

jmz52, 2020-05-20:
Sorry, wrong branch.
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/ultralcd.cpp#L884

GMagician, 2020-05-20:
I found it....I not analyzed how lcd works but it seems you did it and since you found that I think you may post a PR. If in trouble posting it I may do it for you

AnHardt, 2020-05-20:
When i wrote the cod i never assumed a screen update could be as short as below a ms.
`if (max_display_update_time) max_display_update_time--;` seems to be the right solution.
If a screen update does not last longer, the probability the planner-buffer runs dry in that time is extremely low. Wasting a ms by an artificial delay would be wrong.

The line exists to make the algorithm adaptive. A very long one time screen-update (for example interrupted by something) should not have influence forever.

closed 2020-05-20 by thinkyhead

jmz52, 2020-05-22:
@GMagician, thanks for the PR

commit 0b0ba7dc (2020-05-20): Fix LCD throttle issue (#18055)

Fixes #18044