MarlinFirmware/Marlin #18045: [FR] filament real motion sensors as speed limiter in the x and y axes.

p2004r, 2020-05-19:
Cheap and accurate filament real motion sensors are available.
For example https://www.thingiverse.com/thing:2515750

Now the firmware limits print speed because the extruder has the **declared** ability to pull the filament. Why not use the filament extraction speed **actually obtained from the sensor** readings with the optical encoder as this parameter?

In a first step in this way, the firmware can analyze the maximum achievable filament speed during printing (by moments when filament not really moving) and display an estimate of this value. Monitoring the history of this parameter objectively warns the user of problems in the hot end (e.g. the need for cleanup).

ellensp, 2020-05-20:
The firmware limits the speed based on the ability to heat the filament up fast enough. Has nothing to so with the ability to pull the the filament.

p2004r, 2020-05-20:
In addition to the amount of plastic that melts through the hot end, this also limits the ability to push the plastic through the nozzle of the selected diameter.
The nozzle is constantly coking during operation. Its actual diameter is reduced and needs to be cleaned at an unknown (and mostly inappropriate) time.
In addition, the diameter of a new nozzle is not always known exactly.