MarlinFirmware/Marlin #18046: Error compiling Marlin 2.0.x

NameTheJew, 2020-05-19:
Get this error when trying to compile for CR-10

Arduino: 1.8.12 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

c:/users/home/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: C:\Users\home\AppData\Local\Temp\arduino_build_140435/Marlin.ino.elf section `.text' will not fit in region `text'

c:/users/home/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: region `text' overflowed by 708 bytes

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Sanguino.

swilkens, 2020-05-19:
You have enabled too many features to fit on the board's memory, it doesn't fit.

If you want suggestions regarding what you can turn off, please upload your `configuration.h` and `configuration-adv.h`  - as the template asks from you.

NameTheJew, 2020-05-20:
Following a suggestion from Chris Riley (whos [guide ](https://youtu.be/7J7NYnxL5vA) im following) I downgraded to Arduino 1.8.5
The error changes
> Arduino: 1.8.5 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"
> 
> fork/exec C:\Users\home\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5/bin/avr-gcc.exe: The filename or extension is too long.
> Error compiling for board Sanguino.
Verbose error log attached again.
[error verbose 1.8.5.txt](https://github.com/MarlinFirmware/Marlin/files/4654390/error.verbose.1.8.5.txt)

I notice the following error repeats in the log
> Wno-error=narrowing
Would shortening the file name solve this?

> @swilkens You have enabled too many features to fit on the board's memory, it doesn't fit.
I've copied the CR-10 example config straight from the Configurations-release-2.0.5
Surely the default example config should fit? But it doesnt...
The main feature i care about trying is "LIN_ADVANCE"
but i get this error even when it is commented out

Powerprobot, 2020-05-20:
are you able to read?  

> You have enabled too many features to fit on the board's memory, it doesn't fit.

If you want suggestions regarding what you can turn off, please upload your configuration.h and configuration-adv.h - as the template asks from you.

> 
Your ATmega has 8-Bit! for LIN_ADVANCE to run, you need a 32-Bit Board. The 8-Bit Board is overload to full function with LIN_ADVANCE.

For the Error with compiler.... check the length of Sub-Folder names. Windows can't handle filenames longer than 256 Characters.

At least... your errors is not an Marlin ISSUE; please close this Issue Number.

swilkens, 2020-05-21:
> but i get this error even when it is commented out

I took the latest bugfix release, grabbed the default configuration files and compiled the binary, result is this:

> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [========= ]  92.3% (used 15128 bytes from 16384 bytes)
> Flash: [==========]  98.1% (used 127620 bytes from 130048 bytes)

So the standard configuration files certainly work just fine.

I had a look, if you disable these; `SHOW_BOOTSCREEN`, `SHOW_CUSTOM_BOOTSCREEN` and `PIDTEMPBED` or `SPEAKER`  you should have enough program memory available to enable `LIN_ADVANCE`. Your bed will switch to ON/OFF heating, but that should be alright for the CR10 printers.

If you don't care about the SD card, you can also disable `SDSUPPORT`. You can keep `PIDTEMPBED` enabled if you kill SD card support.

This is a configuration problem, not a firmware bug.

boelle, 2020-06-21:
@NameTheJew still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle